acooks/tn40xx-driver #1: 88X3140-FW-R02-06-03.hdr

Phhere, 2018-05-03:
Hello,
do you know where i can get the Firmware for Marvell PHY?

acooks, 2018-05-04:
Tehuti Networks say:

> If you own a TN9210 or TN9710 NIC based on Marvell PHY, you will need to obtain the firmware file in .hdr format prior to compiling this driver from either your NIC manufacturer or from Marvell directly.

I understand that that's not helpful or convenient, but it indicates that there is a licensing problem. Unfortunately the files can't be distributed without Marvell's cooperation, but you can probably find an old copy of the driver somewhere that still includes the files.

You can extract the 88X3140-FW-R02-06-03.hdr from the MV88X3120_phy.h or MV88X3140_phy.h file. Look for version 0.3.6.14.1 or earlier of the driver.

acooks, 2018-06-15:
This answer is moving to the wiki, because it's not an issue I can fix.

closed 2018-06-15 by acooks