acooks/tn40xx-driver #3: Problem with .hdr file versions

MarkRivers, 2018-07-19:
The currrent version of the Readme file says this:
```
For TN9710P (DeviceID 4027) please obtain the file x3310fw_0_3_3_0_9374.hdr
For TN9710Q (DeviceID 4527) please obtain the file e2010fw_0_3_3_0_9374.hdr
If you are missing these exact files, the driver will compile without support for these PHYs
```
However the actual versions of these files in the repository are:
x3310fw_0_2_8_0_8850.hdr
e2010fw_0_2_8_0_8850.hdr

So they are not the latest versions referred to in the Readme.  When I build the driver on Centos 7 it does not find my TN9710P based card.  Is it likely this is because it is using the wrong .hdr files?

sheepman0, 2018-07-20:
What does lspci -nn | grep -i network return? Is it definitely the right PCI ID for the driver?

I also noticed this though but have been unable to obtain the x3310fw_0_3_3_0_9374.hdr file from anywhere... Marvell won't talk to me, and seemingly TrendNet (my card vendor) aren't able to help either bizarrely. If anyone can get their hands on x3310fw_0_3_3_0_9374.hdr it'd be much appreciated if they might share it.

sheepman0, 2018-07-20:
Quick update.. I noticed with my card (that has a DeviceID of 4027) the driver wasn't building with support for it. It also errors if you try `make MV88X3310=YES`. 

If your card has the same device ID try copying the attached file into your build directory and running it again. It is still that old x3310fw_0_2_8_0_8850.hdr version but it might get your card working at least.
[MV88X3310_phy.h.zip](https://github.com/acooks/tn40xx-driver/files/2213763/MV88X3310_phy.h.zip)

sheepman0, 2018-07-20:
Glad it fixed it. I would still like to get my hands on the x3310fw_0_3_3_0_9374.hdr file though if anyone has it!

pheff, 2018-07-20:
Please find attached FW hdr file and release notes. Please submit it to
this github project.

pheff, 2018-07-20:
Seems replying by email and attaching file does not post the file here so posting it again.

[x3310fw_0_3_3_0_9374.tar.gz](https://github.com/acooks/tn40xx-driver/files/2214805/x3310fw_0_3_3_0_9374.tar.gz)

russcook, 2018-09-14:
pheff, you are my hero today! You totally saved my bacon.

acooks, 2018-11-25:
@pheff, could you help obtain a distributable copy of x3310fw_0_3_4_0_9445?

pheff, 2018-11-26:
Hi Andrew,
Yes, please give me a few days to find out what is required.
pat

MurrayData, 2018-11-28:
Thanks very much guys, solved it for me too.

lego12239, 2019-01-23:
The same problem - can't find x3310fw_0_3_4_0_9445.hdr .

lego12239, 2019-01-23:
@acooks, why don't you want to include x3310fw_0_3_3_0_9374.hdr in release branch?

lego12239, 2019-03-19:
Hi. I've got x3310fw_0_3_4_0_9445.hdr :-)

[x3310fw_0_3_4_0_9445.hdr.gz](https://github.com/acooks/tn40xx-driver/files/2982468/x3310fw_0_3_4_0_9445.hdr.gz)

lego12239, 2019-03-19:
Please, add this file into git.

jp-bennett, 2019-10-16:
I would assume the 0.2.8.0 version of the firmware is what was last released under a GPL license. Marvell appears to be requiring an NDA for the changelog at the very least. It's very likely that this project does not have the legal right to distribute newer firmware versions. This is a frustrating but common problem.

acooks, 2020-09-12:
@baruchsiach submitted a patch to mainline kernel to add firmware loading for the 88x3310 phy.

The discussion explains the problem:
* the 88x3310 PHY doesn't work without firmware.
* the PHY loads FW from either SPI flash automatically, or must be loaded over MDIO bus by the driver.
* the TN9170P NIC implementation doesn't have a SPI flash to store the PHY firmware, saving a couple of dollars.
* Marvell, who makes the 88x3310 PHY, won't allow redistribution of the firmware files**
* even if the tn40xx driver is converted to use phylib support in mainline kernel, the 88x3310 firmware distribution problem remains.
* this makes the TN9170P and similar NICs with 88x3310 PHY and lacking SPI flash legally unsupportable in an Open Source driver.

** There is a loophole. Any NIC implementation that uses the SPI flash is also distributing the firmware, albeit in a slightly inconvenient package. I suspect that the firmware can easily be read out of the SPI flash of any NIC that contains it, so anyone who ships the flash ROM, also ships the firmware. However, none of this is going to help to convince the mainline kernel to include this firmware.

[Patch submission with discussion](https://patchwork.ozlabs.org/project/netdev/patch/16e4a15e359012fc485d22c7e413a129029fbd0f.1585676858.git.baruch@tkos.co.il/)

[Later version of the patch](https://www.spinics.net/lists/netdev/msg645238.html), presumably with review comments addressed

baruchsiach, 2020-09-13:
>     * this makes the TN9170P and similar NICs with 88x3310 PHY and lacking SPI flash legally unsupportable in an Open Source driver.

IANAL. But I would just like to note that this statement is not accurate in my opinion. It is perfectly legal to apply the firmware load support patch to Open Source driver, and then distribute the result. The patch code is not derived from any non-GPL code.

I presume it is also legal for OEM vendors to distribute the PHY firmware files stored on devices for software to load at run-time. Otherwise, the hardware is not usable.

hfvogt, 2020-10-09:
by the way, the driver package driver(linux)_teg-10gectx_v1(0.3.6.17).tar.gz offered by Trendnet for the discontinued TEG-10GECTX (Version v1.0R), see under
[https://www.trendnet.com/support/support-detail.asp?prod=310_TEG-10GECTX](https://www.trendnet.com/support/support-detail.asp?prod=310_TEG-10GECTX)
includes the 0.3.4.0 firmware for the 88x3310 phy. There is no mentioning in the driver package though, about copyrights.

That is certainly not a long-term solution. However, the firmwares for the 88x3310 and 88e2010 PHYs are always distributed within the windows driver binaries TN40xxmp_64.sys and TN40xxmp_32.sys. I have written a small extraction program, still a bit quick and dirty, but I am happy to share it if there is interest...

adamisvk, 2021-04-13:
Hello, I have managed to get firmware version 0.3.10.0 from another HW.

[x3310fw_0_3_10_0_10860.hdr.gz](https://ufile.io/zp2schwm)

gitontheball, 2021-04-23:
Hi. I see this is the only page on the entire internet that refers to this issue, so I need to ask anyone: I am in urgent and significant need of the other file referred to here, which is the e2010fw_0_3_4_0_9445.hdr file (the firmware for the 88e2010 PHY). Please, if someone could please post that file here, it would be immeasurably appreciated. Please let me know and thanks so much for any and all help locating this.

> by the way, the driver package driver(linux)_teg-10gectx_v1(0.3.6.17).tar.gz offered by Trendnet for the discontinued TEG-10GECTX (Version v1.0R), see under [...]

hfvogt, 2021-06-27:
Hi gitontheball,

if you download the trendnet windows driver where the link was given above and extract the cab file data2.cab under "Installation Files" e.g. with cabextract, then you will find under "DefaultComponent" the windows driver files *.sys. Compile my little extraction program attached
[read-marvell-fw.c.gz](https://github.com/acooks/tn40xx-driver/files/6722004/read-marvell-fw.c.gz)
and call it with the *.sys file as a parameter. Then you get all firmware files in your current directory as file.hdr.new.
Happy extracting :-)

ziswiler, 2022-03-09:
> Hello, I have managed to get firmware version 0.3.10.0 from another HW.
> 
> [x3310fw_0_3_10_0_10860.hdr.gz](https://ufile.io/zp2schwm)

Looks like that one is no longer available for download. @adamisvk could you please share it again?

sheepman0, 2022-03-10:
@ziswiler here you go:

https://mega.nz/file/Zh0GnQIS#eBB0L16lvBAS6nvvY7sZxkVtjwGsraNJJ0pMwCHEiK4

acooks, 2023-12-27:
> @baruchsiach submitted a patch to mainline kernel to add firmware loading for the 88x3310 phy.
>
> The discussion explains the problem:
>
>     * the 88x3310 PHY doesn't work without firmware.
>
>     * the [...]

@wkz has posted another patch set to add support for loading 88x3310 PHY firmware to the mainline kernel:

https://lore.kernel.org/netdev/20231214201442.660447-1-tobias@waldekranz.com/

In the discussion, the same concerns were raised about where to get the firmware and how to distribute it. Unfortunately little or no progress has been made on that front.