acooks/tn40xx-driver #5: Log spam: "rxd_err = 0x10"

acooks, 2018-08-30:
The tn40xx driver is continuously printing "rxd_err = 0x10" to the kernel log.
The message is logged with error severity and indicates a TCP checksum error.

[178130.688609] tn40xx: loading out-of-tree module taints kernel.
[178130.688919] tn40xx: module verification failed: signature and/or required key missing - tainting kernel
[178130.691125] Tehuti Network Driver, 0.3.6.16.1
[178130.691127] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG 
[178130.691402] srom 0x0 HWver 16 build 0 lane# 4 max_pl 0x1 mrrs 0x2
[178130.927129] PHY detected on port 1 ID=43A400 - QT2025 10Gbps SFP+
[178132.554209] QT2025 FW version 2.0.3.3 module type 0x4
[178132.569387] tn40xx 0000:01:00.0 enp1s0: renamed from eth0
[178132.657703] fw 0xe
[178132.657714] enp1s0, Port A
[178132.657728] 1 1fc9:4022:1fc9:3015
[178132.657747] detected 1 cards, 1 loaded
[178132.678580] IPv6: ADDRCONF(NETDEV_UP): enp1s0: link is not ready
[178132.788944] IPv6: ADDRCONF(NETDEV_UP): enp1s0: link is not ready
[178135.014124] enp1s0 Link Up 10G
[178135.014159] IPv6: ADDRCONF(NETDEV_CHANGE): enp1s0: link becomes ready
[178138.497468] rxd_err = 0x10
[178139.496917] rxd_err = 0x10
[178139.520752] rxd_err = 0x10
[178141.544918] rxd_err = 0x10
[178141.568767] rxd_err = 0x10
[178145.576895] rxd_err = 0x10
[178145.599649] rxd_err = 0x10
[178153.705395] rxd_err = 0x10
[178153.728670] rxd_err = 0x10
[178208.829274] rxd_err = 0x10
[178209.879790] rxd_err = 0x10
[178210.900655] rxd_err = 0x10
[178211.918293] rxd_err = 0x10
[178213.972620] rxd_err = 0x10
[178215.948609] rxd_err = 0x10

acooks, 2019-04-21:
Receive checksum offloading appears to be broken and also permanently enabled.

Using ethtool to disable rx checksums does not result in any error and subsequent queries show that it's disabled, but the driver doesn't support it.

DatPat, 2024-10-20:
> Receive checksum offloading appears to be broken and also permanently enabled.
> 
> Using ethtool to disable rx checksums does not result in any error and subsequent queries show that it's disabled, but the driver doesn't support it.

Do you have any documentation for this chipset or are you basing your work entirely on the released driver?