acooks/tn40xx-driver #9: error: ‘struct ethtool_ops’ has no member named ‘get_settings’; did you mean ‘get_strings’?

Infinality, 2019-01-25:
I'm having compile issues with kernel 4.20.3.  Looks like "get_settings()" has been deprecated.  I'm guessing this is an upstream issue, but does anybody have some quick fix in the meantime?

`$ dkms install -m tn40xx -v 001 -k 4.20.3-200.fc29.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.20.3-200.fc29.x86_64 MV88X3310=YES -C /lib/modules/4.20.3-200.fc29.x86_64/build M=/var/lib/dkms/tn40xx/001/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.20.3-200.fc29.x86_64 (x86_64)
Consult /var/lib/dkms/tn40xx/001/build/make.log for more information.

$ cat /var/lib/dkms/tn40xx/001/build/make.log
DKMS make.log for tn40xx-001 for kernel 4.20.3-200.fc29.x86_64 (x86_64)
Fri Jan 25 12:17:05 CST 2019
make: Entering directory '/usr/src/kernels/4.20.3-200.fc29.x86_64'
  CC [M]  /var/lib/dkms/tn40xx/001/build/tn40.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/CX4.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/CX4_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/MV88X3310_phy.o
/var/lib/dkms/tn40xx/001/build/tn40.c: In function ‘bdx_ethtool_ops’:
/var/lib/dkms/tn40xx/001/build/tn40.c:4937:4: error: ‘struct ethtool_ops’ has no member named ‘get_settings’; did you mean ‘get_strings’?
   .get_settings = bdx_get_settings,
    ^~~~~~~~~~~~
    get_strings
/var/lib/dkms/tn40xx/001/build/tn40.c:4937:19: error: initialization of ‘int (*)(struct net_device *, struct ethtool_fecparam *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_cmd *)’ [-Werror=incompatible-pointer-types]
   .get_settings = bdx_get_settings,
                   ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4937:19: note: (near initialization for ‘bdx_ethtool_ops.get_fecparam’)
/var/lib/dkms/tn40xx/001/build/tn40.c:4938:4: error: ‘struct ethtool_ops’ has no member named ‘set_settings’; did you mean ‘get_strings’?
   .set_settings = bdx_set_settings,
    ^~~~~~~~~~~~
    get_strings
/var/lib/dkms/tn40xx/001/build/tn40.c:4938:19: error: initialization of ‘int (*)(struct net_device *, struct ethtool_fecparam *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_cmd *)’ [-Werror=incompatible-pointer-types]
   .set_settings = bdx_set_settings,
                   ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4938:19: note: (near initialization for ‘bdx_ethtool_ops.set_fecparam’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:291: /var/lib/dkms/tn40xx/001/build/tn40.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1567: _module_/var/lib/dkms/tn40xx/001/build] Error 2
make: Leaving directory '/usr/src/kernels/4.20.3-200.fc29.x86_64'

`

Infinality, 2019-01-25:
Answering my own question.  I was able to get it to compile by commenting out the offending legacy code.   ;)  Patch below.

`
--- tn40.c.ORIG 2019-01-25 12:36:30.822311313 -0600
+++ tn40.c      2019-01-25 15:25:39.042780965 -0600
@@ -4934,8 +4935,10 @@
 #ifdef ETHTOOL_SLINKSETTINGS
                .set_link_ksettings = bdx_set_link_ksettings,
 #endif
+/*
                .get_settings = bdx_get_settings,
                .set_settings = bdx_set_settings,
+*/
                .get_drvinfo = bdx_get_drvinfo,
                .get_link = ethtool_op_get_link,
                .get_coalesce = bdx_get_coalesce,
`

closed 2019-01-25 by Infinality

rayfoss, 2019-04-05:
Removing the offending lines from tn40.c worked to compile the driver for Akitio on Fedora https://www.akitio.com/faq/342-thunder2-10gbe-adapter-can-i-use-this-network-adapter-on-linux

jlittlenz, 2019-05-18:
It worked, too, for compiling the Intel e1000e driver on Ubuntu 19.04.