acooks/tn40xx-driver #10: rxd_err = 0x8

markbza, 2019-01-29:
Hi Friends! Thank you so much on your awesome work on this driver!!

We are running a 10gbit solution with your drivers on multiple machines. Our machines seem to lock up after a few days like clockwork. It starts giving rxd_err = 0x8 about a page full then the machine hard locks.

error is: rxd_err = 0x8
adapter model number is: Edimax EN-9320SFP+
kernel version:  4.15.18-30
driver version = latest (7months ago github)

Please advise what we can try?

Infinality, 2019-01-30:
I don't actually know much about the modifications that acooks did to the drivers, but it might be described in a README somewhere.  If you are using the github version, you might try the stock upstream version, or vice versa.

acooks, 2023-12-17:
overtaken by events

closed 2023-12-17 by acooks