acooks/tn40xx-driver #14: Unable to compile on Fedora 30

therevoman, 2019-06-19:
I've been using your driver on Fedora with minimal changes for the last 6 months.  I was even albe to compile in Fedora 30 until today.   

The first error below is what I am seeing with branch release/tn40xx-002.
Previously, with branch tn40xx-001 I would modify tn40.c by commenting out the definition of .get_settings and .set_settings and everything would compile.   I guess the latest kernel has some changes in the kernel headers.  Any tips are appreciated

```
Your branch is up to date with 'origin/release/tn40xx-002'.

# make all
make -C /lib/modules/5.1.8-300.fc30.x86_64/build M=/opt/projects/github/tn40xx-driver clean
make[1]: Entering directory '/usr/src/kernels/5.1.8-300.fc30.x86_64'
make[1]: Leaving directory '/usr/src/kernels/5.1.8-300.fc30.x86_64'
Building kernel 5.1.8-300.fc30.x86_64 resume supported
make -C /lib/modules/5.1.8-300.fc30.x86_64/build M=/opt/projects/github/tn40xx-driver modules 
make[1]: Entering directory '/usr/src/kernels/5.1.8-300.fc30.x86_64'
  CC [M]  /opt/projects/github/tn40xx-driver/tn40.o
  CC [M]  /opt/projects/github/tn40xx-driver/CX4.o
  CC [M]  /opt/projects/github/tn40xx-driver/CX4_Linux.o
  CC [M]  /opt/projects/github/tn40xx-driver/QT2025_phy.o
  CC [M]  /opt/projects/github/tn40xx-driver/QT2025_phy_Linux.o
  CC [M]  /opt/projects/github/tn40xx-driver/TLK10232_phy.o
  CC [M]  /opt/projects/github/tn40xx-driver/TLK10232_phy_Linux.o
  CC [M]  /opt/projects/github/tn40xx-driver/AQR105_phy.o
  CC [M]  /opt/projects/github/tn40xx-driver/AQR105_phy_Linux.o
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_set_link_mode’:
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
   92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
   99 |  if (speed == AQR105_ALL_SPEEDS) {
      |               ^~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:92:28: note: each undeclared identifier is reported only once for each function it appears in
   92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
   99 |  if (speed == AQR105_ALL_SPEEDS) {
      |               ^~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_get_link_ksettings’:
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
   92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:133:50: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
  133 |  AQR105_set_link_mode(cmd->link_modes.supported, AQR105_ALL_SPEEDS);
      |                                                  ^~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_set_link_ksettings’:
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
   92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/projects/github/tn40xx-driver/AQR105_phy_Linux.c:158:48: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
  158 |   AQR105_set_link_mode(priv->link_advertising, AQR105_ALL_SPEEDS);
      |                                                ^~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:276: /opt/projects/github/tn40xx-driver/AQR105_phy_Linux.o] Error 1
make[1]: *** [Makefile:1575: _module_/opt/projects/github/tn40xx-driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.1.8-300.fc30.x86_64'
make: *** [Makefile:147: all] Error 2

```

Infinality, 2019-06-21:
I think I figured it out.   Change this line:
#define MV88X3310_ALL_SPEEDS    (__ETHTOOL_LINK_MODE_LAST+1)

To this:
#define MV88X3310_ALL_SPEEDS    (__ETHTOOL_LINK_MODE_MASK_NBITS+1)

Compiles now.

therevoman, 2019-10-10:
That seems to work.  How did you know MODE_MASK_NBITS was equivalent to MODE_LAST?

Infinality, 2019-10-18:
I'm pretty sure that the compiler suggested it.

jcheger, 2019-12-25:
in Linux 4.x headers, "**__ETHTOOL_LINK_MODE_MASK_NBITS**" is defined as "(__ETHTOOL_LINK_MODE_LAST+1)".

The correct fix is "__ETHTOOL_LINK_MODE_MASK_NBITS", but without "+1".

Full explanation in issue #18

acooks, 2020-01-08:
Fixed in release/tn40xx-003.

closed 2020-01-08 by acooks