acooks/tn40xx-driver #15: Unable to build on Debian 10 (OpenMediaVault)

Hoerli1337, 2019-10-22:
Hello, I'm only getting this as a result:
`root@openmediavault:/usr/src/tn40xx-002# dkms install -m tn40xx -v 002

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=5.2.0-0.bpo.3-amd64 -C /lib/modules/5.2.0-0.bpo.3-amd64/build M=/var/lib/dkms/tn40xx/002/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.2.0-0.bpo.3-amd64 (x86_64)
Consult /var/lib/dkms/tn40xx/002/build/make.log for more information.
`
I followed the instructions.
https://github.com/acooks/tn40xx-driver/blob/release/tn40xx-002/docs/dkms.md
Even with this great manual, which once worked for Debian 9, it doesn't work anymore.
https://noobient.com/2018/08/10/edimax-en-9320tx-e-10gbe-ethernet-card-on-ubuntu/

>> Debian 10
>> OpenMediaVaul 5.0.10-1

GPSLeo, 2019-10-24:
I have the same issue on Ubuntu 19.10 with 5.3.0-19-generic Kernel.
Part of the make.log showing the issue:

> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c: In function ‘AQR105_set_link_mode’:
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
>    92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
>       |                            ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
>    99 |  if (speed == AQR105_ALL_SPEEDS) {
>       |               ^~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:92:28: note: each undeclared identifier is reported only once for each function it appears in
>    92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
>       |                            ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
>    99 |  if (speed == AQR105_ALL_SPEEDS) {
>       |               ^~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c: In function ‘AQR105_get_link_ksettings’:
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
>    92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
>       |                            ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:133:50: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
>   133 |  AQR105_set_link_mode(cmd->link_modes.supported, AQR105_ALL_SPEEDS);
>       |                                                  ^~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c: In function ‘AQR105_set_link_ksettings’:
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
>    92 | #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
>       |                            ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/002/build/AQR105_phy_Linux.c:158:48: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
>   158 |   AQR105_set_link_mode(priv->link_advertising, AQR105_ALL_SPEEDS);
>       |                                                ^~~~~~~~~~~~~~~~~
>

jcheger, 2019-12-25:
in Linux 4.x headers, "**__ETHTOOL_LINK_MODE_MASK_NBITS**" is defined as "(__ETHTOOL_LINK_MODE_LAST+1)".

The correct fix is "__ETHTOOL_LINK_MODE_MASK_NBITS", but without "+1".

Full explanation in issue #18

acooks, 2020-01-08:
This is fixed in release/tn40xx-003.

closed 2020-01-08 by acooks