acooks/tn40xx-driver #17: PTP disabled

thecaptury, 2019-11-29:
Hi,

is there a reason that PTP support is disabled by default?

In the header tn40.h uncommenting TN40_PTP allows me to use PTP and it doesn't appear to have any negative side effects...

acooks, 2020-01-08:
Thanks for the report!
I don't know why the PTP support is disabled by default, but I also don't have a way to test it. Could you provide any more information about the environment you tested in? Kernel version, hardware device, and any other test equipment would provide a valuable reference point.

thecaptury, 2020-01-08:
I tested on stock kernels on Debian 9, Debian 10, Ubuntu 16.04, Ubuntu 18.04, and Ubuntu 19.10.

I also closely examined the source code. I don't see anything that could go wrong. It essentially just publishes the timestamp when a packet is pushed to the hardware to the meta data of the packet.