acooks/tn40xx-driver #18: kernel 5 fix

jcheger, 2019-12-25:
tn40xx - Linux kernel 5 fix
===========================
Tn40xx driver does not compile against Linux kernel 5.x.

procedure for reproducing this fix
----------------------------------
```sh
git clone https://github.com/acooks/tn40xx-driver.git
cd tn40xx-driver
sed -ie 's/__ETHTOOL_LINK_MODE_LAST+1/__ETHTOOL_LINK_MODE_MASK_NBITS/' *_Linux.c
```

Acooks driver
-------------
This patch will be proposed as a fix to [Acooks](https://github.com/acooks/tn40xx-driver). If you did download it,
it's not mandatory to download this repository instead of the original one. You only have to patch the files.

### [DKMS way](https://github.com/acooks/tn40xx-driver/blob/release/tn40xx-002/docs/dkms.md)
```sh
cd /usr/src/tn40xx-002
sed -ie 's/__ETHTOOL_LINK_MODE_LAST+1/__ETHTOOL_LINK_MODE_MASK_NBITS/' *_Linux.c
dkms install -m tn40xx -v 002
```

### manual compile way
```sh
cd wherever you did download the driver
sed -ie 's/__ETHTOOL_LINK_MODE_LAST+1/__ETHTOOL_LINK_MODE_MASK_NBITS/' *_Linux.c
make
```

explanation
-----------
Kernel 4.x headers include the file "ethtool.h", that contains:
```c
/* number of link mode bits/ulongs handled internally by kernel */
#define __ETHTOOL_LINK_MODE_MASK_NBITS                  \
        (__ETHTOOL_LINK_MODE_LAST + 1)
```

**__ETHTOOL_LINK_MODE_LAST** is not defined anymore in kernel 5.x, but **__ETHTOOL_LINK_MODE_MASK_NBITS** still exists.

Replacing "**__ETHTOOL_LINK_MODE_LAST+1**" by "**__ETHTOOL_LINK_MODE_MASK_NBITS**" solves the problem.

Files modified:
* AQR105_phy_Linux.c
* MV88X3120_phy_Linux.c
* MV88X3310_phy_Linux.c

compile error
-------------
```make
make -C /lib/modules/5.3.13-1-pve/build M=/root/tn40xx-driver clean
make[1]: Entering directory '/usr/src/linux-headers-5.3.13-1-pve'
make[1]: Leaving directory '/usr/src/linux-headers-5.3.13-1-pve'
Building kernel 5.3.13-1-pve resume supported
make -C /lib/modules/5.3.13-1-pve/build M=/root/tn40xx-driver modules 
make[1]: Entering directory '/usr/src/linux-headers-5.3.13-1-pve'
  CC [M]  /root/tn40xx-driver/tn40.o
  CC [M]  /root/tn40xx-driver/CX4.o
  CC [M]  /root/tn40xx-driver/CX4_Linux.o
  CC [M]  /root/tn40xx-driver/QT2025_phy.o
  CC [M]  /root/tn40xx-driver/QT2025_phy_Linux.o
  CC [M]  /root/tn40xx-driver/TLK10232_phy.o
  CC [M]  /root/tn40xx-driver/TLK10232_phy_Linux.o
  CC [M]  /root/tn40xx-driver/AQR105_phy.o
  CC [M]  /root/tn40xx-driver/AQR105_phy_Linux.o
/root/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_set_link_mode’:
/root/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
 #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
                            ^~~~~~~~~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
  if (speed == AQR105_ALL_SPEEDS) {
               ^~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c:92:28: note: each undeclared identifier is reported only once for each function it appears in
 #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
                            ^~~~~~~~~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c:99:15: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
  if (speed == AQR105_ALL_SPEEDS) {
               ^~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_get_link_ksettings’:
/root/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
 #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
                            ^~~~~~~~~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c:133:50: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
  AQR105_set_link_mode(cmd->link_modes.supported, AQR105_ALL_SPEEDS);
                                                  ^~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c: In function ‘AQR105_set_link_ksettings’:
/root/tn40xx-driver/AQR105_phy_Linux.c:92:28: error: ‘__ETHTOOL_LINK_MODE_LAST’ undeclared (first use in this function); did you mean ‘__ETHTOOL_LINK_MODE_MASK_NBITS’?
 #define AQR105_ALL_SPEEDS (__ETHTOOL_LINK_MODE_LAST+1)
                            ^~~~~~~~~~~~~~~~~~~~~~~~
/root/tn40xx-driver/AQR105_phy_Linux.c:158:48: note: in expansion of macro ‘AQR105_ALL_SPEEDS’
   AQR105_set_link_mode(priv->link_advertising, AQR105_ALL_SPEEDS);
                                                ^~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:290: /root/tn40xx-driver/AQR105_phy_Linux.o] Error 1
make[1]: *** [Makefile:1655: _module_/root/tn40xx-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.13-1-pve'
make: *** [Makefile:147: all] Error 2
```

formtapez, 2020-01-03:
In version [0.3.6.17.2](http://www.tehutinetworks.net/images/UL240756/tn40xx-0.3.6.17.2.tgz) (vendor-download since may 2019) they fixed this issue similar - by replacing the value by 0xFFFFFFFF.

acooks, 2020-01-08:
Thanks very much for the investigation, solution and report, @jcheger and @formtapez

The 0xFFFFFFFF replacement in the 0.3.6.17.2 vendor drop is not the preferred solution. I decided to revert the replacement and implement the solution suggested here, because it matches the intended upstream change.

This fix is in the release/tn40xx-003 branch.

closed 2020-01-08 by acooks