acooks/tn40xx-driver #20: Tehuti Networks out of business - driver forking and maintenance

acooks, 2020-01-09:
Tehuti Networks [posted the following notice on LinkedIn](https://www.linkedin.com/posts/tehuti-networks_after-16-years-tehuti-networks-is-closing-activity-6548469995167334400-npiK):

> After 16 years, Tehuti Networks is closing its doors.
> 
> I had the privilege of serving as the company's COO for the last 9 years in which we have made 
> extraordinary achievements in bringing 10GbE to the Consumer market.
> Unfortunately, the market did not grow at the expected rate and we had to make the tough call.
> 
> I'd like to thank our investors, Proseed Venture Capital Fund Ltd. and especially Mr. Ligad Rotlevy, Mr. Moshe Shahaf, Mr. Niv Sarne and Mr. Shai Levy for providing me with this opportunity and believing in what we do throughout these years.
> I'd like to thank the small, amazing team of employees working hard to make this happen.
> I'd like to thank our customers, partners and vendors for taking this journey with us.
> 
> It's been a hell of a ride!
> Thank you all!
> Nir Sever.

In light of this announcement, I do not expect to see ongoing development or maintenance releases from Tehuti. Effectively, there is no more upstream vendor for this driver.

On the positive side, this makes it much easier to refactor and develop the driver without having to prepare for future upstream changes. Going forward, the development workflow will be much simpler, with a more standard branching and merging model.

Leave a comment if you have strong feelings about the future of this driver.

slyrus, 2020-09-02:
Appreciate your work on the driver. Thanks!

kalvdans, 2020-11-13:
My company have just added a "1fc9:4025 Tehuti Networks Ltd. TN9510 10GBase-T/NBASE-T Ethernet Adapter" pci card our product, so we will have resources to continue maintaining the driver.

pierrextardif, 2021-06-01:
Hello,

Thanks for the great work on the drivers.

I'm using a `TN9210 10G ethernet Adapter`.
kernel version : `5.8.0-53-generic`

do you know where i can find the drivers for it?

@kalvdans maybe you havew a working version?

Thks,

P

pierrextardif, 2021-06-02:
So `TN9210` drivers can be installed on kernel `5.8.0-53-generic` by downloading the header `MV88X3120_phy.h` from an [older release](https://github.com/acooks/tn40xx-driver/blob/release/tn40xx-001/MV88X3120_phy.h) , then running `make MV88X3120=YES` .
Thanks for your great work and support on this.

SummerSeaSun, 2022-06-07:
Manufacturer like Startech.com discontinued products and provide driver for old kernel only, no support.
I'm on a Debian Stable and current kernel is 5.10.0-13-amd64, thanks for your work on the drivers, but is there anybody here that follow the project right now?