acooks/tn40xx-driver #26: drive doesnt compile at 5.4.x Kernel

spooch99, 2020-02-18:
Hi,
Ive tried to compile the Tehuti 40xx-drive for a newer 5.4.x-Kernel, but unfortunatly it didnt work. Ive attached the compiler log for seeing, what happened. It seems probably a compiler issue of a newer ggc. The System is an Ubuntu 18.04.4 64bit Server Installation.

Tfh!
Oliver
[compile-log.txt](https://github.com/acooks/tn40xx-driver/files/4218866/compile-log.txt)

acooks, 2020-02-19:
The error log shows that the (ubuntu provided) compiler is unhappy about the (ubuntu provided) kernel headers. It looks like this system is not able to compile any kernel modules at all.

acooks, 2020-09-02:
@spooch99 if there's anything more to add to this issue, please reopen.

closed 2020-09-02 by acooks

spooch99, 2020-09-02:
Hi Andrew,

sorry for my late reply! The code now compiles without problems (Ubuntu
20.04 and Kernel 5.8.5).

Regards,
Oliver