acooks/tn40xx-driver #29: Driver in kernel >5.4 possible? Tainted kernel 5.7

raeschenzentrum, 2020-05-25:
Driver compiles in kernel 5.7 but while loading is tainting the kernel. Is it even possible to use the Driver with kernel > 5.4?

Hoerli1337, 2020-08-15:
Same with kernel 5.6

`root@debian10:/usr/src# dkms add -m tn40xx -v 003

Creating symlink /var/lib/dkms/tn40xx/003/source ->
                 /usr/src/tn40xx-003

DKMS: add completed.
root@debian10:/usr/src# dkms install -m tn40xx -v 003

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j2 KERNELRELEASE=5.6.0-0.bpo.2-amd64 -C /lib/modules/5.6.0-0.bpo.2-amd64/build M=/var/lib/dkms/tn40xx/003/build........
cleaning build area...

DKMS: build completed.

tn40xx.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.6.0-0.bpo.2-amd64/updates/dkms/

depmod....

DKMS: install completed.`

modprobe tn40xx delivers absolutely no output.

acooks, 2020-09-02:
The new release/tn40xx-004 branch includes the fix for kernel 5.7 and supports 5.4 and newer.
Tested on Fedora 31 with 5.7.15-100.

closed 2020-09-02 by acooks

Hoerli1337, 2020-09-04:
Installing the driver is now possible with kernel 5.7.
At least I don't get an error message.
But the network card does not appear in the system itself and there is no link to the switch.

# lspci
02:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter

# modprobe tn40xx
nothing

Am I doing something wrong?
I have installed everything normally via dkms.

Debian 10
Linux 5.7.0-0.bpo.2-amd64 #1 SMP Debian 5.7.10-1~bpo10+1 (2020-07-30) x86_64 GNU/Linux