acooks/tn40xx-driver #31: Crash with Linux 5.7 - WARNING: CPU: 6 PID: 644 at net/ethtool/common.c:348 ethtool_check_ops+0x14/0x20

presslab-us, 2020-07-08:
I had this working fine on Linux 5.6, but today my PC upgraded to 5.7 and the tn40xx driver crashes.

```
0.000000] Linux version 5.7.7-200.fc32.x86_64 (mockbuild@bkernel01.iad2.fedoraproject.org) (gcc version 10.1.1 20200507 (Red Hat 10.1.1-1) (GCC), GNU ld version 2.34-3.fc32) #1 SMP Wed Jul 1 19:53:01 UTC 2020
...
[   51.697332] tn40xx: loading out-of-tree module taints kernel.
[   51.697389] tn40xx: module verification failed: signature and/or required key missing - tainting kernel
[   51.697661] Tehuti Network Driver, 0.3.6.17.2
[   51.697662] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG 
[   51.697704] tn40xx 0000:01:00.0: enabling device (0000 -> 0002)
[   51.697816] srom 0x0 HWver 16 build 0 lane# 4 max_pl 0x1 mrrs 0x2
[   51.929042] PHY detected on port 1 ID=3A1B4A3 - AQR105 10Gbps 10GBase-T
...
[   58.553169] AQR105 FW ver: 2.c.72
[   58.575344] ------------[ cut here ]------------
[   58.575348] WARNING: CPU: 6 PID: 644 at net/ethtool/common.c:348 ethtool_check_ops+0x14/0x20
[   58.575348] Modules linked in: intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_sof_pci snd_sof_intel_byt kvm snd_sof_intel_ipc iwlmvm(+) snd_sof_intel_hda_common snd_soc_hdac_hda irqbypass snd_sof_xtensa_dsp intel_cstate snd_sof_intel_hda mac80211 snd_sof intel_rapl_perf snd_soc_skl snd_hda_codec_hdmi snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi libarc4 snd_hda_codec_realtek wmi_bmof snd_soc_core iwlwifi intel_wmi_thunderbolt snd_hda_codec_generic snd_compress ledtrig_audio ac97_bus snd_pcm_dmaengine cfg80211 pcspkr mxm_wmi snd_hda_intel snd_intel_dspcfg snd_usb_audio(+) snd_hda_codec snd_usbmidi_lib snd_hda_core snd_rawmidi snd_hwdep i2c_i801 thunderbolt snd_seq snd_seq_device btusb uvcvideo snd_pcm btrtl btbcm btintel videobuf2_vmalloc bluetooth snd_timer videobuf2_memops videobuf2_v4l2 snd videobuf2_common ecdh_generic videodev rfkill soundcore powermate joydev mc ecc i2c_multi_instantiate tn40xx(OE+) acpi_pad
[   58.575363]  acpi_tad mei_me mei intel_pch_thermal ip_tables hid_logitech_hidpp hid_logitech_dj uas usb_storage i915 crct10dif_pclmul i2c_algo_bit cec crc32_pclmul crc32c_intel drm_kms_helper nvme drm r8169 nvme_core ghash_clmulni_intel wmi video fuse
[   58.575368] CPU: 6 PID: 644 Comm: systemd-udevd Tainted: G           OE     5.7.7-200.fc32.x86_64 #1
[   58.575368] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./Z490 Phantom Gaming-ITX/TB3, BIOS P1.30 05/14/2020
[   58.575369] RIP: 0010:ethtool_check_ops+0x14/0x20
[   58.575370] Code: bc f4 ff ff ff eb e6 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 0f 1f 44 00 00 31 c0 48 83 7f 70 00 74 0d 8b 17 85 d2 75 07 <0f> 0b b8 ea ff ff ff c3 0f 1f 40 00 0f 1f 44 00 00 4c 8b 8f f8 01
[   58.575371] RSP: 0018:ffffa58a40573ba8 EFLAGS: 00010246
[   58.575372] RAX: 0000000000000000 RBX: ffffffffb99e3000 RCX: ffff8f6ad89e7df0
[   58.575372] RDX: 0000000000000000 RSI: 000000000000003e RDI: ffffffffc0967000
[   58.575372] RBP: ffff8f6ade7ef000 R08: 0000000000000000 R09: 000000000000c432
[   58.575373] R10: 0000000000000000 R11: ffffa58a406a0000 R12: 0000000000000001
[   58.575373] R13: ffffa58a402b9000 R14: ffff8f6ade7ef900 R15: ffff8f6ae25950b0
[   58.575374] FS:  00007fcd5ccdeb80(0000) GS:ffff8f6ae5580000(0000) knlGS:0000000000000000
[   58.575374] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   58.575375] CR2: 00007fef14537000 CR3: 0000000461bdc004 CR4: 00000000003606e0
[   58.575375] Call Trace:
[   58.575377]  register_netdevice+0x59/0x630
[   58.575380]  ? _cond_resched+0x16/0x40
[   58.575381]  register_netdev+0x1c/0x40
[   58.575384]  bdx_probe+0x264/0x990 [tn40xx]
[   58.575387]  local_pci_probe+0x42/0x80
[   58.575388]  ? _cond_resched+0x16/0x40
[   58.575389]  pci_device_probe+0xd9/0x190
[   58.575390]  really_probe+0x167/0x410
[   58.575391]  driver_probe_device+0xb6/0x100
[   58.575392]  device_driver_attach+0xa1/0xb0
[   58.575393]  __driver_attach+0x8a/0x150
[   58.575394]  ? device_driver_attach+0xb0/0xb0
[   58.575394]  ? device_driver_attach+0xb0/0xb0
[   58.575395]  bus_for_each_dev+0x64/0x90
[   58.575396]  bus_add_driver+0x12b/0x1e0
[   58.575397]  driver_register+0x8b/0xe0
[   58.575399]  ? bdx_tx_timeout+0x30/0x30 [tn40xx]
[   58.575400]  do_one_initcall+0x46/0x200
[   58.575401]  ? _cond_resched+0x16/0x40
[   58.575403]  ? kmem_cache_alloc_trace+0x167/0x220
[   58.575404]  ? do_init_module+0x23/0x260
[   58.575405]  do_init_module+0x5c/0x260
[   58.575406]  __do_sys_init_module+0x14f/0x170
[   58.575407]  do_syscall_64+0x5b/0xf0
[   58.575408]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   58.575409] RIP: 0033:0x7fcd5de3040e
[   58.575410] Code: 48 8b 0d 8d 0a 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 5a 0a 0c 00 f7 d8 64 89 01 48
[   58.575411] RSP: 002b:00007fff130a6eb8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[   58.575411] RAX: ffffffffffffffda RBX: 000055fa83b37160 RCX: 00007fcd5de3040e
[   58.575412] RDX: 00007fcd5da8a95d RSI: 000000000008ae38 RDI: 000055fa83b534c0
[   58.575412] RBP: 000055fa83b534c0 R08: 000055fa83b31010 R09: 00007fff130a5b98
[   58.575413] R10: 0000000000000002 R11: 0000000000000246 R12: 0000000000000000
[   58.575413] R13: 00007fcd5da8a95d R14: 000055fa83b31080 R15: 000055fa83b314e0
[   58.575414] ---[ end trace 4041894c56d7b050 ]---
[   58.575415] register_netdev failed
[   58.575420] ------------[ cut here ]------------
[   58.575421] WARNING: CPU: 6 PID: 644 at net/core/dev.c:8802 rollback_registered_many+0x9f/0x490
[   58.575421] Modules linked in: intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_sof_pci snd_sof_intel_byt kvm snd_sof_intel_ipc iwlmvm(+) snd_sof_intel_hda_common snd_soc_hdac_hda irqbypass snd_sof_xtensa_dsp intel_cstate snd_sof_intel_hda mac80211 snd_sof intel_rapl_perf snd_soc_skl snd_hda_codec_hdmi snd_soc_sst_ipc snd_soc_sst_dsp snd_hda_ext_core snd_soc_acpi_intel_match snd_soc_acpi libarc4 snd_hda_codec_realtek wmi_bmof snd_soc_core iwlwifi intel_wmi_thunderbolt snd_hda_codec_generic snd_compress ledtrig_audio ac97_bus snd_pcm_dmaengine cfg80211 pcspkr mxm_wmi snd_hda_intel snd_intel_dspcfg snd_usb_audio(+) snd_hda_codec snd_usbmidi_lib snd_hda_core snd_rawmidi snd_hwdep i2c_i801 thunderbolt snd_seq snd_seq_device btusb uvcvideo snd_pcm btrtl btbcm btintel videobuf2_vmalloc bluetooth snd_timer videobuf2_memops videobuf2_v4l2 snd videobuf2_common ecdh_generic videodev rfkill soundcore powermate joydev mc ecc i2c_multi_instantiate tn40xx(OE+) acpi_pad
[... 61 lines cut ...]
[   58.584064]  __driver_attach+0x8a/0x150
[   58.584064]  ? device_driver_attach+0xb0/0xb0
[   58.584065]  ? device_driver_attach+0xb0/0xb0
[   58.584066]  bus_for_each_dev+0x64/0x90
[   58.584066]  bus_add_driver+0x12b/0x1e0
[   58.584067]  driver_register+0x8b/0xe0
[   58.584070]  ? bdx_tx_timeout+0x30/0x30 [tn40xx]
[   58.584071]  do_one_initcall+0x46/0x200
[   58.584072]  ? _cond_resched+0x16/0x40
[   58.584073]  ? kmem_cache_alloc_trace+0x167/0x220
[   58.584074]  ? do_init_module+0x23/0x260
[   58.584075]  do_init_module+0x5c/0x260
[   58.584076]  __do_sys_init_module+0x14f/0x170
[   58.584078]  do_syscall_64+0x5b/0xf0
[   58.584079]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[   58.584080] RIP: 0033:0x7fcd5de3040e
[   58.584081] Code: 48 8b 0d 8d 0a 0c 00 f7 d8 64 89 01 48 83 c8 ff c3 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 49 89 ca b8 af 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 5a 0a 0c 00 f7 d8 64 89 01 48
[   58.584082] RSP: 002b:00007fff130a6eb8 EFLAGS: 00000246 ORIG_RAX: 00000000000000af
[   58.584083] RAX: ffffffffffffffda RBX: 000055fa83b37160 RCX: 00007fcd5de3040e
[   58.584083] RDX: 00007fcd5da8a95d RSI: 000000000008ae38 RDI: 000055fa83b534c0
[   58.584083] RBP: 000055fa83b534c0 R08: 000055fa83b31010 R09: 00007fff130a5b98
[   58.584084] R10: 0000000000000002 R11: 0000000000000246 R12: 0000000000000000
[   58.584084] R13: 00007fcd5da8a95d R14: 000055fa83b31080 R15: 000055fa83b314e0
[   58.591165] tn40xx: probe of 0000:01:00.0 failed with error -22
```

hogend, 2020-07-09:
Same for me:

TN9710P with Marvel PHY

```
Linux kelewan 5.7.0-1-amd64 #1 SMP Debian 5.7.6-1 (2020-06-24) x86_64 GNU/Linux
....
[    7.722332] MV88X3310 initdata applied
[    7.722415] MV88X3310 I/D version is 0.3.4.0
[    7.822699] ------------[ cut here ]------------
[    7.822708] WARNING: CPU: 1 PID: 338 at net/ethtool/common.c:346 ethtool_check_ops+0x14/0x20
[    7.822709] Modules linked in: intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi nls_ascii snd_hda_intel snd_virtuoso nls_cp437 snd_intel_dspcfg snd_oxygen_lib snd_hda_codec vfat kvm_intel snd_mpu401_uart fat snd_hda_core snd_hwdep snd_pcm_oss kvm snd_mixer_oss irqbypass cuse ghash_clmulni_intel fuse snd_pcm nvidia_drm(POE) aesni_intel drm_kms_helper libaes crypto_simd cec cryptd snd_seq_midi glue_helper nvidia_modeset(POE) evdev snd_seq_midi_event mei_wdt snd_rawmidi nvidia(POE) intel_cstate snd_seq intel_uncore ipmi_devintf snd_seq_device intel_rapl_perf ipmi_msghandler snd_timer nct6775 hwmon_vid coretemp snd iTCO_wdt 8021q garp stp sg mrp llc iTCO_vendor_support mei_me soundcore mei watchdog tn40xx(OE+) pcspkr button loop nfsd auth_rpcgss nfs_acl lockd drm grace sunrpc ip_tables x_tables autofs4 xfs raid10 raid1 raid0 multipath linear dm_mirror dm_region_hash dm_log dm_mod raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor
[    7.822762]  raid6_pq libcrc32c crc32c_generic md_mod sd_mod t10_pi crc_t10dif crct10dif_generic ahci libahci libata crct10dif_pclmul crct10dif_common i2c_i801 crc32_pclmul crc32c_intel xhci_pci xhci_hcd scsi_mod ehci_pci ehci_hcd lpc_ich mfd_core e1000e usbcore ptp pps_core usb_common video
[    7.822780] CPU: 1 PID: 338 Comm: systemd-udevd Tainted: P           OE     5.7.0-1-amd64 #1 Debian 5.7.6-1
[    7.822782] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./B85M Pro4, BIOS P1.90 12/06/2013
[    7.822785] RIP: 0010:ethtool_check_ops+0x14/0x20
[    7.822789] Code: bc f4 ff ff ff eb e6 66 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 40 00 0f 1f 44 00 00 31 c0 48 83 7f 70 00 74 0d 8b 17 85 d2 75 07 <0f> 0b b8 ea ff ff ff c3 0f 1f 40 00 0f 1f 44 00 00 4c 8b 8f f8 01
[    7.822790] RSP: 0018:ffffb443404d7a68 EFLAGS: 00010246
[    7.822793] RAX: 0000000000000000 RBX: ffff88a40c17a000 RCX: ffff88a4020fec10
[    7.822794] RDX: 0000000000000000 RSI: ffffb44340661270 RDI: ffffffffc0b57000
[    7.822796] RBP: ffff88a407c56000 R08: 0000000000000000 R09: 0000000000000000
[    7.822797] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[    7.822798] R13: ffffffff866f3c80 R14: ffff88a407c56000 R15: ffff88a40c17a0b0
[    7.822801] FS:  00007f839bf2e880(0000) GS:ffff88a40dc80000(0000) knlGS:0000000000000000
[    7.822802] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    7.822804] CR2: 00007fe357de7de0 CR3: 000000040b592006 CR4: 00000000001606e0
[    7.822806] Call Trace:
[    7.822814]  register_netdevice+0x6a/0x570
[    7.822827]  ? MV88X3310_mdio_reset.cold+0x3b7/0x5e3 [tn40xx]
[    7.822832]  register_netdev+0x1c/0x40
[    7.822841]  bdx_probe+0x290/0x930 [tn40xx]
[    7.822850]  local_pci_probe+0x42/0x80
[    7.822855]  pci_device_probe+0x104/0x1a0
[    7.822861]  really_probe+0x147/0x3c0
[    7.822864]  driver_probe_device+0xb6/0x100
[    7.822868]  device_driver_attach+0x53/0x60
[    7.822871]  __driver_attach+0x8a/0x150
[    7.822874]  ? device_driver_attach+0x60/0x60
[    7.822877]  ? device_driver_attach+0x60/0x60
[    7.822880]  bus_for_each_dev+0x78/0xc0
[    7.822883]  bus_add_driver+0x14d/0x1f0
[    7.822887]  driver_register+0x6c/0xc0
[    7.822894]  ? bdx_tx_timeout+0x30/0x30 [tn40xx]
[    7.822898]  do_one_initcall+0x46/0x200
[    7.822903]  ? free_unref_page_commit+0x8b/0x110
[    7.822909]  ? _cond_resched+0x15/0x30
[    7.822914]  ? kmem_cache_alloc_trace+0x161/0x220
[    7.822917]  ? do_init_module+0x23/0x230
[    7.822921]  do_init_module+0x5c/0x230
[    7.822924]  load_module+0x234c/0x24f0
[    7.822931]  ? __do_sys_finit_module+0xaa/0x110
[    7.822933]  __do_sys_finit_module+0xaa/0x110
[    7.822940]  do_syscall_64+0x52/0x180
[    7.822944]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[    7.822947] RIP: 0033:0x7f839c49af79
[    7.822950] Code: 00 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 48 89 f8 48 89 f7 48 89 d6 48 89 ca 4d 89 c2 4d 89 c8 4c 8b 4c 24 08 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d e7 3e 0c 00 f7 d8 64 89 01 48
[    7.822952] RSP: 002b:00007ffd6e9da5a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    7.822954] RAX: ffffffffffffffda RBX: 00005583f28c4f40 RCX: 00007f839c49af79
[    7.822956] RDX: 0000000000000000 RSI: 00007f839c39be4d RDI: 0000000000000006
[    7.822957] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    7.822958] R10: 0000000000000006 R11: 0000000000000246 R12: 00007f839c39be4d
[    7.822960] R13: 0000000000000000 R14: 00005583f28bbea0 R15: 00005583f28c4f40
[    7.822963] ---[ end trace 9ac3b3609560fa5f ]---
[    7.822966] register_netdev failed
[    7.823333] ------------[ cut here ]------------
[    7.823338] WARNING: CPU: 1 PID: 338 at net/core/dev.c:8800 rollback_registered_many+0xa9/0x530
[    7.823339] Modules linked in: intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp snd_hda_codec_hdmi nls_ascii snd_hda_intel snd_virtuoso nls_cp437 snd_intel_dspcfg snd_oxygen_lib snd_hda_codec vfat kvm_intel snd_mpu401_uart fat snd_hda_core snd_hwdep snd_pcm_oss kvm snd_mixer_oss irqbypass cuse ghash_clmulni_intel fuse snd_pcm nvidia_drm(POE) ae [... cut ...]
[    7.830913] RSP: 002b:00007ffd6e9da5a8 EFLAGS: 00000246 ORIG_RAX: 0000000000000139
[    7.830916] RAX: ffffffffffffffda RBX: 00005583f28c4f40 RCX: 00007f839c49af79
[    7.830918] RDX: 0000000000000000 RSI: 00007f839c39be4d RDI: 0000000000000006
[    7.830919] RBP: 0000000000020000 R08: 0000000000000000 R09: 0000000000000000
[    7.830921] R10: 0000000000000006 R11: 0000000000000246 R12: 00007f839c39be4d
[    7.830922] R13: 0000000000000000 R14: 00005583f28bbea0 R15: 00005583f28c4f40
[    7.850425] tn40xx: probe of 0000:05:00.0 failed with error -22
```

slyrus, 2020-07-24:
I've got basically the same issue here trying to upgrade from 5.6 to 5.7.

cahz, 2020-08-07:
Does the mentioned PR fix the problem for you?

presslab-us, 2020-08-07:
Yes!  Thanks dude.  Up and running on Fedora with 5.7.11-200.fc32.x86_64.

acooks, 2020-09-02:
Thanks for the fix, @cahz !
The new release/tn40xx-004 branch supports 5.4 to 5.7 and includes the fix.
Tested on Fedora 31 with 5.7.15-100.

closed 2020-09-02 by acooks

presslab-us, 2020-09-02:
@acooks thanks for continuing to maintain this!