acooks/tn40xx-driver #36: Makefile does not respect KERNELRELEASE passed by DKMS

skwerlman, 2020-12-16:
When building the module with DKMS, the Makefile assumes that we only want to build the module for the current kernel, even if we tell DKMS that we want a different one with the `-k` option. On arch linux at least, this results in the DKMS install phase of a kernel upgrade failing, and a subsequent loss of networking on reboot.

DKMS tries to run the Makefile with the following command:
```
make -j32 KERNELRELEASE=5.9.14-zen1-1-zen -C /usr/lib/modules/5.9.14-zen1-1-zen/build M=/var/lib/dkms/tn40xx/004/build
```

The makefile ignores the KERNELRELEASE option, and uses `uname -r` instead, which results in the following error:
```
DKMS make.log for tn40xx-004 for kernel 5.9.14-zen1-1-zen (x86_64)
Tue Dec 15 10:29:14 PM EST 2020
make: Entering directory '/usr/lib/modules/5.9.14-zen1-1-zen/build'
/var/lib/dkms/tn40xx/004/build/Makefile:140: *** Aborting the build: Linux kernel /lib/modules/5.9.13-zen1-1-zen/build source not found.  Stop.
make: *** [Makefile:1781: /var/lib/dkms/tn40xx/004/build] Error 2
make: Leaving directory '/usr/lib/modules/5.9.14-zen1-1-zen/build'
```

kalvdans, 2020-12-16:
Does https://github.com/acooks/tn40xx-driver/pull/34 fix this?

skwerlman, 2020-12-17:
yes, #34 does fix this, thanks

acooks, 2023-12-16:
Fixed in release/tn40xx-006

closed 2023-12-16 by acooks