acooks/tn40xx-driver #41: Debian11  ST10GSPEXNDP not working

mlc42, 2021-04-06:
I´m trying to install the driver under kernel 5.10.0.5. 
dkms runs without a problem, but no NIC comes up.
The same with the patched sources from  christsich

lscpi -k shows 2 TN9710P tied to kernelmodules tn40xx.

after modprobe dmesg shows

tn40xx: probe of 0000:04:00.0 failed with errror -22
tn40xx: PHY detected on Port 0 ID=2B09AB - MV88X3310 (A1) 10 Gps 10GBase-T
tn40xx: MV88X3310  initdata applied
tn40xx: MV88X3310 I/D version is 0.2.8.0

hofkosk, 2021-04-08:
Hello... Same issue there. Debian testing + kernel 5.10.0.5.

mlc42, 2021-04-10:
i wonder how this changes : KERNEL_VERSION(5, 4, 0)  to KERNEL_VERSION(5, 10, 0) could solve that.
#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 4, 0) has same result as KERNEL_VERSION(5, 10, 0)

i´ve tried this but no luck. dmesg shows tn40xx: register_netdevice failed 
May be there is other magic behind. 
Do you have a  ST10GSPEXNDP and Debian Buster (or Bullseye). I´v tried all with Bullseye Version because
only with the nonfree version i can get the MSI Onboard NIC running.

I assume you use the startech-patched version from christasich.

hofkosk, 2021-04-13:
Sorry for my late answer, but I can't simulate start position. I have installed Bullseye with PEX10000SFP on another SSD and after netinstall with Cinnamon i run for correct boot with AMD RX 6800 XT :

apt install git dkms build-essential module-assistant linux-headers-$(uname -r) nfs-common autofs
mkdir ~/git
cd ~/git
git clone https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
cp -r linux-firmware/amdgpu /lib/firmware
update-initramfs -u -k all
apt install firmware-amd-graphics libgl1-mesa-dri libglx-mesa0 mesa-vulkan-drivers xserver-xorg-video-all
shutdown -r now

and start testing with attached driver

cd /home/drivers/tehuti/
m-a prepare
make
make install
modprobe tn40xx

card not working. After this I try install drivers via acooks - dkms and not working. After this I try only rmmod tn40xx and modprobe tn40xx and card been detected and working and work after restart too. I don't have any idea why... This procedure can help you as well maybe.
[5.10.0-5-amd64.tar.gz](https://github.com/acooks/tn40xx-driver/files/6306697/5.10.0-5-amd64.tar.gz)

mlc42, 2021-04-14:
thank you for your answer,

i´ve tried this files but no luck.

awilson, 2021-07-08:
I have the same issue with a recent upgrade to Debian11.  See the output from dmesg attached.

I've also tried the suggestions above to no avail.

[dmesg.Errors.txt](https://github.com/acooks/tn40xx-driver/files/6787015/dmesg.Errors.txt)

mikevipe, 2022-02-14:
I got this working on Debian 11 5.10.0-11, please find my build files attached, I could not get either acooks or christasich version to work out of the box so I just downloaded acooks, and copied christasich's tn40.c, MV88X3310_phy.c, AQR105_phy.c and added x3310fw_0_3_4_0_9445.hdr, ran: make uninstall, make clean, make, make install did a reboot and the device appeared. I am not sure why combining them worked but it did.
 :
[tn40xx-driver-release-tn40xx-004-02-14-2022-edited.zip](https://github.com/acooks/tn40xx-driver/files/8060167/tn40xx-driver-release-tn40xx-004-02-14-2022-edited.zip)

rnaheed, 2022-02-20:
Verified that the above build by mikevipe does indeed work on Debian 11 on kernel 5.10.0-11-amd64 .  The symptoms with the original driver were that the driver would be loaded with modprobe, but not create any network devices.  With the above build, it does.

rnaheed, 2022-02-24:
Unfortunately had to back the driver out due to kernel panics (out of memory) two nights in a row.  So there may be a memory leak in the 02-14-2022 driver.

SummerSeaSun, 2022-05-26:
This is the output of dmesg for modprobe tn40xx:

```
[  926.774137] perf: interrupt took too long (2506 > 2500), lowering kernel.perf_event_max_sample_rate to 79750
[ 2941.474499] perf: interrupt took too long (3135 > 3132), lowering kernel.perf_event_max_sample_rate to 63750
[ 4802.210342] Tehuti Network Driver from https://github.com/acooks/tn40xx-driver, 004
[ 4802.210346] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG 

```

** modinfo tn40xx**
```
filename:       /lib/modules/5.10.0-13-amd64/updates/dkms/tn40xx.ko
description:    Tehuti Network Driver from https://github.com/acooks/tn40xx-driver
author:         Tehuti networks
version:        004
license:        GPL
srcversion:     5E745B89103BF5F5C3B0C06
alias:          pci:v0000105Ad00007203sv0000105Asd00007203bc*sc*i*
alias:          pci:v00001FC9d00004025sv00001432sd00008102bc*sc*i*
alias:          pci:v00001FC9d00004025sv00001FC9sd00003015bc*sc*i*
alias:          pci:v00001FC9d00004025sv00001186sd00002900bc*sc*i*
alias:          pci:v00001FC9d00004026sv00004C52sd00001000bc*sc*i*
alias:          pci:v00001FC9d00004026sv00001FC9sd00003015bc*sc*i*
alias:          pci:v00001FC9d00004022sv00001432sd00008103bc*sc*i*
alias:          pci:v00001FC9d00004022sv00001043sd00008709bc*sc*i*
alias:          pci:v00001FC9d00004022sv00001186sd00004D00bc*sc*i*
alias:          pci:v00001FC9d00004022sv00001FC9sd00003015bc*sc*i*
alias:          pci:v00001FC9d00004020sv0000180Csd00002040bc*sc*i*
alias:          pci:v00001FC9d00004020sv00001FC9sd00003015bc*sc*i*
alias:          pci:v00001FC9d00004010sv00001FC9sd00004010bc*sc*i*
depends:        
retpoline:      Y
name:           tn40xx
vermagic:       5.10.0-13-amd64 SMP mod_unload modversions 
parm:           no_phy:int
parm:           bdx_force_no_phy_mode:no_phy=1 - force no phy mode (CX4)

```

I can see the hardware in lshw:
```

description: PCI bridge
             product: 5520/5500/X58 I/O Hub PCI Express Root Port 7
             vendor: Intel Corporation
             physical id: 7
             bus info: pci@0000:00:07.0
             version: 22
             width: 32 bits
             clock: 33MHz
             capabilities: pci msi pciexpress pm normal_decode bus_master cap_list
             configuration: driver=pcieport
             resources: irq:26 ioport:efefe00000(size=1048576)
           *-network
                description: Ethernet controller
                product: TN9710P 10GBase-T/NBASE-T Ethernet Adapter
                vendor: Tehuti Networks Ltd.
                physical id: 0
                bus info: pci@0000:03:00.0
                logical name: /dev/fb0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: msi pm pciexpress bus_master cap_list fb
                configuration: depth=32 latency=0 mode=1024x768 visual=truecolor xres=1024 yres=768
                resources: iomemory:ef0-eef memory:efefe00000-efefe0ffff
```

But cannot configure the interface.
BR

SummerSeaSun, 2022-06-08:
dmesg  | grep tn40

```

[    6.170566] tn40xx: loading out-of-tree module taints kernel.
[    6.171300] tn40xx: module verification failed: signature and/or required key missing - tainting kernel
[    6.173417] tn40xx: Tehuti Network Driver, 0.3.6.17.1
[    6.173467] tn40xx: Supported phys : MV88X3120 MV88X3310  QT2025 TLK10232 AQR105 MUSTANG 
[    6.175610] tn40xx: srom 0x0 HWver 16 build 0 lane# 4 max_pl 0x1 mrrs 0x2
[    6.411402] tn40xx: PHY detected on port 0 ID=2B09AB - MV88X3310 (A1) 10Gbps 10GBase-T
[    9.554787] tn40xx: MV88X3310 initdata applied
[    9.555592] tn40xx: MV88X3310 I/D version is 0.2.8.0
[    9.643238] Modules linked in: nls_utf8 cifs libarc4 dns_resolver overlay fscache libdes intel_powerclamp ipmi_ssif coretemp bonding kvm_intel kvm irqbypass ghash_clmulni_intel aesni_intel libaes crypto_simd cryptd glue_helper intel_cstate intel_uncore pcspkr sg iTCO_wdt cp210x intel_pmc_bxt iTCO_vendor_support watchdog joydev usbserial ioatdma acpi_ipmi tn40xx(OE+) dca i5500_temp ipmi_si i7core_edac ipmi_devintf ipmi_msghandler evdev acpi_cpufreq fuse configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 raid0 multipath linear md_mod hid_generic usbhid hid sr_mod sd_mod cdrom t10_pi crc_t10dif crct10dif_generic ahci mgag200 i2c_algo_bit libahci drm_kms_helper libata cec uhci_hcd ehci_pci scsi_mod ehci_hcd drm crct10dif_pclmul crct10dif_common crc32_pclmul e1000e usbcore crc32c_intel i2c_i801 i2c_smbus lpc_ich ptp usb_common pps_core button
[    9.643333]  ? MV88X3310_mdio_reset.cold+0x343/0x6f2 [tn40xx]
[    9.643341]  bdx_probe+0x2c8/0xb10 [tn40xx]
[    9.643377]  ? bdx_tx_timeout+0x20/0x20 [tn40xx]
[    9.643413] tn40xx: register_netdev failed
[    9.644119] Modules linked in: nls_utf8 cifs libarc4 dns_resolver overlay fscache libdes intel_powerclamp ipmi_ssif coretemp bonding kvm_intel kvm irqbypass ghash_clmulni_intel aesni_intel libaes crypto_simd cryptd glue_helper intel_cstate intel_uncore pcspkr sg iTCO_wdt cp210x intel_pmc_bxt iTCO_vendor_support watchdog joydev usbserial ioatdma acpi_ipmi tn40xx(OE+) dca i5500_temp ipmi_si i7core_edac ipmi_devintf ipmi_msghandler evdev acpi_cpufreq fuse configfs ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c crc32c_generic raid1 raid0 multipath linear md_mod hid_generic usbhid hid sr_mod sd_mod cdrom t10_pi crc_t10dif crct10dif_generic ahci mgag200 i2c_algo_bit libahci drm_kms_helper libata cec uhci_hcd ehci_pci scsi_mod ehci_hcd drm crct10dif_pclmul crct10dif_common crc32_pclmul e1000e usbcore crc32c_intel i2c_i801 i2c_smbus lpc_ich ptp usb_common pps_core button
[    9.644200]  bdx_probe+0x352/0xb10 [tn40xx]
[    9.644230]  ? bdx_tx_timeout+0x20/0x20 [tn40xx]
[    9.723487]  bdx_probe+0x352/0xb10 [tn40xx]
[    9.723525]  ? bdx_tx_timeout+0x20/0x20 [tn40xx]
[    9.766891] tn40xx: probe of 0000:03:00.0 failed with error -22

```

mikevipe, 2022-06-08:
I know you are running Ubuntu vs my Debian 11 but have you tried to use the files I linked with kernel 5.10.0 on Ubuntu? It is worth a shot. You can use the mainline kernel installer to install 5.10.0. Or follow the guide at https://ubuntuhandbook.org/index.php/2020/12/install-linux-kernel-5-10-ubuntu-linux-mint/

As for running on Debian, I have done this with 4 machines now all with success. On Debian you MUST use the 5.10.0-11 kernel for it to work. All of my routers are running the Intel(R) Xeon(R) CPU E3-1270 v3 @ 3.50GHz with 32GB ram and I have not yet had a memory leak issue or crash with the version I submitted. This router is doing about 2.3gbps down and 1.73gbps up consistently. 

root@router-h2:~# uname -a
Linux router-h2 5.10.0-11-amd64 #1 SMP Debian 5.10.92-1 (2022-01-18) x86_64 GNU/Linux
root@router-h2:~# free -m
               total        used        free      shared  buff/cache   available
Mem:           32051        1287       29686           9        1077       30376
Swap:           7622           0        7622
root@router-h2:~# uptime
 03:02:02 up 102 days,  1:26,  0 users,  load average: 4.49, 4.67, 4.00
root@router-h2:~#

Please let me know if it works with Ubuntu on the 5.10.0 kernel.

SummerSeaSun, 2022-06-13:
Thank you @mikevipe, I'm using Debian too not Ubuntu.
Kernel is: Linux  5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64.
CPU is:  Intel(R) Xeon(R) E5645  @ 2.40GHz with 16GB RAM.

I must downgrade kernel or it could be that ram is too low to correctly run?

Manozoys1, 2024-06-24:
If anyone facing the following issue during build time, please use the updated patch file,

=================================
Ubuntu: Ubuntu 20.04
Kernel Version: 5.15.0-67-generic
=================================

sudo make all
make -C /lib/modules/5.15.0-67-generic/build M=/home/Mano/tn40xx-driver-release-tn40xx-004 clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-67-generic'
  CLEAN   /home/Mano/tn40xx-driver-release-tn40xx-004/Module.symvers
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-67-generic'
Building kernel 5.15.0-67-generic resume supported
make -C /lib/modules/5.15.0-67-generic/build M=/home/Mano/tn40xx-driver-release-tn40xx-004 modules 
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-67-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
  You are using:           gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
  CC [M]  /home/Mano/tn40xx-driver-release-tn40xx-004/tn40.o
/home/Mano/tn40xx-driver-release-tn40xx-004/tn40.c: In function ‘bdx_ethtool_ops’:
/home/Mano/tn40xx-driver-release-tn40xx-004/tn40.c:4097:19: **error:** initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ **[-Werror=incompatible-pointer-types]**
 4097 |   .get_coalesce = **bdx_get_coalesce,**
      |                   ^~~~~~~~~~~~~~~~
/home/Mano/tn40xx-driver-release-tn40xx-004/tn40.c:4097:19: note: (near initialization for ‘bdx_ethtool_ops.get_coalesce’)
/home/Mano/tn40xx-driver-release-tn40xx-004/tn40.c:4098:19: **error:** initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ **[-Werror=incompatible-pointer-types]**
 4098 |   .set_coalesce = **bdx_set_coalesce,**
      |                   ^~~~~~~~~~~~~~~~
/home/Mano/tn40xx-driver-release-tn40xx-004/tn40.c:4098:19: note: (near initialization for ‘bdx_ethtool_ops.set_coalesce’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/Mano/tn40xx-driver-release-tn40xx-004/tn40.o] Error 1
make[1]: *** [Makefile:1906: /home/Mano/tn40xx-driver-release-tn40xx-004] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-67-generic'
make: *** [Makefile:147: all] Error 2

**Updated Patch:**

[tn40xx-driver-release-tn40xx-004-06-24-2024.zip](https://github.com/user-attachments/files/15951406/tn40xx-driver-release-tn40xx-004-06-24-2024.zip)

Manozoys1, 2025-05-21:
=========================================================
### Ubuntu: Ubuntu 24.04.2
Kernel Version: 6.8.0-51-generic

**Updated Patch:**
[

[tehuti_driver_ubuntu24.zip](https://github.com/user-attachments/files/20365355/tehuti_driver_ubuntu24.zip)

](url)