acooks/tn40xx-driver #45: ST-Lab 480 (TN9710P) not detected after DKMS install

WhiteFossa, 2021-06-05:
Hi All,

I've just bought a pair of ST-Lab N-480 10G adapters and installed them into machines with Fedora 34.

`uname -a
Linux amphimachairodus 5.12.8-300.fc34.x86_64 #1 SMP Fri May 28 15:20:54 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
`
NICs are present in my lspci:
`05:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter`

I've cloned repo and installed DKMS as described here https://github.com/acooks/tn40xx-driver without any problems.

Then rebooted - an don't see NIC neither in ifconfig -a, nor in dmesg:
`dmesg | grep tn4` returns just empty set.

Any suggestions, what can be missing?

WhiteFossa, 2021-06-05:
After manually inserting module:
tn40xx: loading out-of-tree module taints kernel.
[ 1603.735364] tn40xx: module verification failed: signature and/or required key missing - tainting kernel
[ 1603.735989] Tehuti Network Driver from https://github.com/acooks/tn40xx-driver, 004
[ 1603.735991] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG

I.e. there is a problem with my PHY (which is Marvell's).

WhiteFossa, 2021-06-05:
If I add firmware files, then I'm getting compilation error, related to damaged MV88X3310_PHY_H:

#ifndef _MV88X3310_PHY_H
#define _MV88X3310_PHY_H

static u16 MV88X3310_phy_initdata[] __initdata = {
/* x3310fw_0_3_4_0_9445.hdr */
unsigned int MV88X3310_phy_initdata_len = 
#endif

This is bullshit from that file

randomsamples, 2021-07-05:
Im also unable to get my card working which is a Trendnet TEG-10GECTX. Driver builds, appears to install, wont detect my card. I manually modprobe it in and it looks very similar to what youre hitting. Would love to know how I can troubleshoot this.

makz27, 2021-10-21:
I think i found the issue, since dkms change the make folder with the option -C, the Makefile can't find the .hdr file and so do not build the support of MV88X3310

I've removed the option in dkms.conf

MAKE[0]="make M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build"

and now the module is build with the supported card.