acooks/tn40xx-driver #46: is there possible to update this driver to support latest linux kernel like 5.11+?

afrojewelz, 2021-08-16:
is there possible to update this driver to support latest linux kernel like 5.11+?
i ve tried use dkms install on proxmox which use 5.11+ kernel not work
i ve tried make && make install still not work 
tried serveral branch 003、004、0.36.17.2

TerminalAddict, 2021-08-16:
I've got a new proxmox
5.11.22-3-pve
aptitude install pve-header-`uname -r`
git clone -b release/tn40xx-004 https://github.com/acooks/tn40xx-driver.git /usr/src/tn40xx-004
dkms add -m tn40xx -v 004
dkms install -m tn40xx -v 004
dkms remove -m tn40xx -v 003 --all

afrojewelz, 2021-08-17:
> I've got a new proxmox
> 5.11.22-3-pve
> aptitude install pve-headers-5.11.22-3-pve
> git clone -b release/tn40xx-004 https://github.com/acooks/tn40xx-driver.git /usr/src/tn40xx-004
> dkms add -m tn40xx -v 004
> dkms install -m tn40xx -v 004
> dkms remove -m tn40xx -v 003 --all

does it work? i have tried this and failed

TerminalAddict, 2021-08-17:
Yep sure does.

afrojewelz, 2021-08-17:
but after i dkms install this module,it does not autoload to kernel or the device not loaded from demsg or lsmod|grep
even i modprobe it,it just spell out two lines in dmesg the device wouldn't open

TerminalAddict, 2021-08-17:
Then something about your setup is different.
But I literally built a Debian 11 proxmox today with this driver.

afrojewelz, 2021-08-17:
03:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961/SM963 [144d:a804]
        Subsystem: Samsung Electronics Co Ltd SM963 2.5" NVMe PCIe SSD [144d:a801]
        Kernel driver in use: nvme
04:00.0 Ethernet controller [0200]: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter [1fc9:4027]
        Subsystem: Tehuti Networks Ltd. Ethernet Adapter [1fc9:3015]
05:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961/SM963 [144d:a804]
        Subsystem: Samsung Electronics Co Ltd SM963 2.5" NVMe PCIe SSD [144d:a801]
        Kernel driver in use: nvme
06:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM961/PM961/SM963 
it apears the kernel module is load by manual when i nano it into /etc/modules,but still device wouldn't open and interact with kernel