acooks/tn40xx-driver #47: Fails to build against kernel 5.15.2-zen1-1-zen on arch linux

skwerlman, 2021-11-15:
This broke when the ethtool coalesce api was extended in https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f3ccfda1931977b80267ba54070a1aeafa18f6ca

DKMS output:
```
[root@a3 sk]# dkms install tn40xx/004 -k 5.15.2-zen1-1-zen
Deprecated feature: REMAKE_INITRD

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.15.2-zen1-1-zen -C /usr/lib/modules/5.15.2-zen1-1-zen/build M=/var/lib/dkms/tn40xx/004/build KVERSION=5.15.2-zen1-1-zen...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.15.2-zen1-1-zen (x86_64)
Consult /var/lib/dkms/tn40xx/004/build/make.log for more information.
```
`/var/lib/dkms/tn40xx/004/build/make.log`:
```log
DKMS make.log for tn40xx-004 for kernel 5.15.2-zen1-1-zen (x86_64)
Mon Nov 15 12:29:15 PM EST 2021
make: Entering directory '/usr/lib/modules/5.15.2-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/tn40xx/004/build/tn40.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy_Linux.o
/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_ethtool_ops’:
/var/lib/dkms/tn40xx/004/build/tn40.c:4097:33: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 4097 |                 .get_coalesce = bdx_get_coalesce,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4097:33: note: (near initialization for ‘bdx_ethtool_ops.get_coalesce’)
/var/lib/dkms/tn40xx/004/build/tn40.c:4098:33: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 4098 |                 .set_coalesce = bdx_set_coalesce,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4098:33: note: (near initialization for ‘bdx_ethtool_ops.set_coalesce’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:277: /var/lib/dkms/tn40xx/004/build/tn40.o] Error 1
make: *** [Makefile:1863: /var/lib/dkms/tn40xx/004/build] Error 2
make: Leaving directory '/usr/lib/modules/5.15.2-zen1-1-zen/build'
```

quick patch which fixed it for me:
```diff
diff --git a/mnt/code/tn40xx-driver/tn40.c b/usr/src/tn40xx-004/tn40.c
index 51d1f06..0e80582 100644
--- a/mnt/code/tn40xx-driver/tn40.c
+++ b/usr/src/tn40xx-004/tn40.c
@@ -3771,7 +3771,7 @@ bdx_get_drvinfo(struct net_device *netdev, struct ethtool_drvinfo *drvinfo)
  * @ecoal
  */
 static int
-bdx_get_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal)
+bdx_get_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *kernel_coal, struct netlink_ext_ack *extack)
 {
        u32 rdintcm;
        u32 tdintcm;
@@ -3803,7 +3803,7 @@ bdx_get_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal)
  * @ecoal
  */
 static int
-bdx_set_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal)
+bdx_set_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *kernel_coal, struct netlink_ext_ack *extack)
 {
        u32 rdintcm;
        u32 tdintcm;
```

a proper patch would need to have docs for the new params, and have version guards

skwerlman, 2022-04-02:
even with the above patch, this is failing again on 5.17.1-zen1-1-zen:

```
DKMS make.log for tn40xx-004 for kernel 5.17.1-zen1-1-zen (x86_64)
Sat Apr  2 01:18:16 PM EDT 2022
make: Entering directory '/usr/lib/modules/5.17.1-zen1-1-zen/build'
  CC [M]  /var/lib/dkms/tn40xx/004/build/tn40.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy_Linux.o
/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_set_mac’:
/var/lib/dkms/tn40xx/004/build/tn40.c:1586:20: warning: passing argument 1 of ‘memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1586 |         memcpy(ndev->dev_addr, addr->sa_data, ndev->addr_len);
      |                ~~~~^~~~~~~~~~
In file included from ./include/linux/string.h:253,
                 from ./include/linux/bitmap.h:11,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/cpumask.h:5,
                 from ./arch/x86/include/asm/msr.h:11,
                 from ./arch/x86/include/asm/processor.h:22,
                 from ./arch/x86/include/asm/timex.h:5,
                 from ./include/linux/timex.h:65,
                 from ./include/linux/time32.h:13,
                 from ./include/linux/time.h:60,
                 from ./include/linux/stat.h:19,
                 from ./include/linux/module.h:13,
                 from /var/lib/dkms/tn40xx/004/build/tn40.h:14,
                 from /var/lib/dkms/tn40xx/004/build/tn40.c:9:
./include/linux/fortify-string.h:212:37: note: expected ‘void *’ but argument is of type ‘const unsigned char *’
  212 | __FORTIFY_INLINE void *memcpy(void *p, const void *q, __kernel_size_t size)
      |                               ~~~~~~^
/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_read_mac’:
/var/lib/dkms/tn40xx/004/build/tn40.c:1602:49: error: assignment of read-only location ‘*(priv->ndev->dev_addr + (sizetype)((int)i * 2 + 1))’
 1602 |                 priv->ndev->dev_addr[i * 2 + 1] = macAddress[i];
      |                                                 ^
/var/lib/dkms/tn40xx/004/build/tn40.c:1603:45: error: assignment of read-only location ‘*(priv->ndev->dev_addr + (sizetype)((int)i * 2))’
 1603 |                 priv->ndev->dev_addr[i * 2] = macAddress[i] >> 8;
      |                                             ^
/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_ethtool_ops’:
/var/lib/dkms/tn40xx/004/build/tn40.c:4099:34: error: initialization of ‘void (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ from incompatible pointer type ‘void (*)(struct net_device *, struct ethtool_ringparam *)’ [-Werror=incompatible-pointer-types]
 4099 |                 .get_ringparam = bdx_get_ringparam,
      |                                  ^~~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4099:34: note: (near initialization for ‘bdx_ethtool_ops.get_ringparam’)
/var/lib/dkms/tn40xx/004/build/tn40.c:4100:34: error: initialization of ‘int (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_ringparam *)’ [-Werror=incompatible-pointer-types]
 4100 |                 .set_ringparam = bdx_set_ringparam,
      |                                  ^~~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4100:34: note: (near initialization for ‘bdx_ethtool_ops.set_ringparam’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:288: /var/lib/dkms/tn40xx/004/build/tn40.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1826: /var/lib/dkms/tn40xx/004/build] Error 2
make: Leaving directory '/usr/lib/modules/5.17.1-zen1-1-zen/build'
```

eli-goldberg-ch, 2022-04-28:
This is also happening for me. 

```
DKMS make.log for tn40xx-001 for kernel 5.15.0-27-generic (x86_64)
Thu Apr 28 11:12:27 PM UTC 2022
make: Entering directory '/usr/src/linux-headers-5.15.0-27-generic'
  CC [M]  /var/lib/dkms/tn40xx/001/build/tn40.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/CX4.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/CX4_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/QT2025_phy.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/QT2025_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/TLK10232_phy.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/TLK10232_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/001/build/AQR105_phy.o
/var/lib/dkms/tn40xx/001/build/tn40.c: In function ‘bdx_tx_map_skb’:
/var/lib/dkms/tn40xx/001/build/tn40.c:3314:30: error: assignment to ‘struct skb_frag_struct *’ from incompatible pointer type ‘skb_frag_t *’ {aka ‘struct bio_vec *’} [-Werror=incompatible-pointer-types]
 3314 |                         frag = &skb_shinfo(skb)->frags[i];
      |                              ^
/var/lib/dkms/tn40xx/001/build/tn40.c:3323:50: error: invalid use of undefined type ‘struct skb_frag_struct’
 3323 |                                              frag->size, PCI_DMA_TODEVICE);
      |                                                  ^~
/var/lib/dkms/tn40xx/001/build/tn40.c:3322:64: error: passing argument 2 of ‘skb_frag_dma_map’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 3322 |                             skb_frag_dma_map(&priv->pdev->dev, frag, 0,
      |                                                                ^~~~
      |                                                                |
      |                                                                struct skb_frag_struct *
In file included from ./include/net/net_namespace.h:39,
                 from ./include/linux/netdevice.h:37,
                 from /var/lib/dkms/tn40xx/001/build/tn40.h:16,
                 from /var/lib/dkms/tn40xx/001/build/tn40.c:9:
./include/linux/skbuff.h:3220:61: note: expected ‘const skb_frag_t *’ {aka ‘const struct bio_vec *’} but argument is of type ‘struct skb_frag_struct *’
 3220 |                                           const skb_frag_t *frag,
      |                                           ~~~~~~~~~~~~~~~~~~^~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:3326:54: error: invalid use of undefined type ‘struct skb_frag_struct’
 3326 |                         bdx_setTxdb(db, dmaAddr, frag->size);
      |                                                      ^~
/var/lib/dkms/tn40xx/001/build/tn40.c: In function ‘bdx_ethtool_ops’:
/var/lib/dkms/tn40xx/001/build/tn40.c:4905:18: error: ‘struct ethtool_ops’ has no member named ‘get_settings’; did you mean ‘get_strings’?
 4905 |                 .get_settings = bdx_get_settings,
      |                  ^~~~~~~~~~~~
      |                  get_strings
/var/lib/dkms/tn40xx/001/build/tn40.c:4905:33: error: initialization of ‘void (*)(struct net_device *, struct ethtool_fec_stats *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_cmd *)’ [-Werror=incompatible-pointer-types]
 4905 |                 .get_settings = bdx_get_settings,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4905:33: note: (near initialization for ‘bdx_ethtool_ops.get_fec_stats’)
/var/lib/dkms/tn40xx/001/build/tn40.c:4906:18: error: ‘struct ethtool_ops’ has no member named ‘set_settings’; did you mean ‘get_strings’?
 4906 |                 .set_settings = bdx_set_settings,
      |                  ^~~~~~~~~~~~
      |                  get_strings
/var/lib/dkms/tn40xx/001/build/tn40.c:4906:33: error: initialization of ‘int (*)(struct net_device *, struct ethtool_fecparam *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_cmd *)’ [-Werror=incompatible-pointer-types]
 4906 |                 .set_settings = bdx_set_settings,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4906:33: note: (near initialization for ‘bdx_ethtool_ops.get_fecparam’)
/var/lib/dkms/tn40xx/001/build/tn40.c:4909:33: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 4909 |                 .get_coalesce = bdx_get_coalesce,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4909:33: note: (near initialization for ‘bdx_ethtool_ops.get_coalesce’)
/var/lib/dkms/tn40xx/001/build/tn40.c:4910:33: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_coalesce *)’ [-Werror=incompatible-pointer-types]
 4910 |                 .set_coalesce = bdx_set_coalesce,
      |                                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/001/build/tn40.c:4910:33: note: (near initialization for ‘bdx_ethtool_ops.set_coalesce’)
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:285: /var/lib/dkms/tn40xx/001/build/tn40.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1875: /var/lib/dkms/tn40xx/001/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.0-27-generic'

```

skwerlman, 2022-04-29:
i have a fork working on the latest kernel here: https://github.com/skwerlman/tn40xx-driver

note that it wont work for some older kernels because im lazy

afrojewelz, 2022-06-05:
i ve managed successfully build tn40xx.ko on proxmox 5.15 for my QNAP Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter [1fc9:4027] 
i will seek methods to share onmy repository with another branch

devadattas, 2022-09-10:
> i ve managed successfully build tn40xx.ko on proxmox 5.15 for my QNAP Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter [1fc9:4027] and finally modprobe attached succesefully i will share methods how to onmy repository https://github.com/afrojewelz/tn40xx-driver/tree/tn40xx-003kvgt515 with another branch，this may only work for my NIC and i may not bare liability to broken your hardware.so if you still wanna try it, proceseds with cautions and thoughts

Thanks this worked for my issue https://github.com/acooks/tn40xx-driver/issues/53 as well.

oktay89, 2023-05-25:
I still have the issue. Already tried a few different repos. 
https://github.com/worksasintended/tn40xx-driver
https://github.com/afrojewelz/tn40xx-driver
My Card is Delock recognized as TN9510 RJ45 10gbe and i am trying to install on proxmox 7.4.3. 
Can you help?

cahz, 2023-05-25:
Maybe try https://github.com/cahz/tn40xx-driver on ´develop` branch

cahz, 2023-05-26:
Kernel 5.17 changed the `ethtool_ops` struct. Maybe try to checkout an older version (like https://github.com/cahz/tn40xx-driver/commit/562b5969dad176fec93652da02a7796b982111e9) which still uses the old struct.

afrojewelz, 2023-08-21:
since my QNAP qm2-2P10G1T card had to continue served,i update my repo branch tn40xx-003kvgt515 into tn40xx-qm2-2P10G1T also make it my default.
https://github.com/afrojewelz/tn40xx-driver/tree/tn40xx-qm2-2P10G1T it' based on tn40xx-003 with header of marvel mv88x3310-phy header , added [cahz] tweakin for dkms ,many thx.