acooks/tn40xx-driver #50: ethtool coalesce compiling problem on all kernels

SientyHD, 2022-07-20:
Hello there,

I keep getting this problem on every linux OS I try:

make: Entering directory '/usr/src/linux-headers-5.15.30-2-pve'
  CC [M]  /var/lib/dkms/tn40xx/004/build/tn40.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy.o
/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_ethtool_ops’:
/var/lib/dkms/tn40xx/004/build/tn40.c:4097:19: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible > 4097 |   .get_coalesce = bdx_get_coalesce,
      |                   ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4097:19: note: (near initialization for ‘bdx_ethtool_ops.get_coalesce’)
/var/lib/dkms/tn40xx/004/build/tn40.c:4098:19: error: initialization of ‘int (*)(struct net_device *, struct ethtool_coalesce *, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)’ from incompatible > 4098 |   .set_coalesce = bdx_set_coalesce,
      |                   ^~~~~~~~~~~~~~~~
/var/lib/dkms/tn40xx/004/build/tn40.c:4098:19: note: (near initialization for ‘bdx_ethtool_ops.set_coalesce’)
  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy_Linux.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy.o
  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy_Linux.o
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:285: /var/lib/dkms/tn40xx/004/build/tn40.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1875: /var/lib/dkms/tn40xx/004/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.30-2-pve'

I cannot figure out what I am doing wrong, I have a Edimax card with Tehuti Networks Ltd. TN9310 10GbE SFP+ Ethernet Adapter

cahz, 2022-07-20:
Did you have a look at PR #48? For even newer Linux kernels, please try b11e0302cf0dbe1d699af0fb1a35a7b38004e767.

cichy45, 2022-08-13:
Hello @cahz 

I tried solutions mentioned by you but it still fails to make. Details: Ubuntu 20.04
uname -r
5.15.0-46-generic
02:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter
(Edimax EN-9320TX-E)

make log:

> DKMS make.log for tn40xx-004 for kernel 5.15.0-46-generic (x86_64)
> nie, 14 sie 2022, 00:39:51 CEST
> make: Wejście do katalogu '/usr/src/linux-headers-5.15.0-46-generic'
>   CC [M]  /var/lib/dkms/tn40xx/004/build/tn40.o
>   CC [M]  /var/lib/dkms/tn40xx/004/build/CX4.o
>   CC [M]  /var/lib/dkms/tn40xx/004/build/CX4_Linux.o
>   CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy.o
> /var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_get_coalesce’:
> /var/lib/dkms/tn40xx/004/build/tn40.c:3780:77: error: parameter name omitted
>  3780 | bdx_get_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)
>       |                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c:3780:111: error: parameter name omitted
>  3780 | bdx_get_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)
>       |                                                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_set_coalesce’:
> /var/lib/dkms/tn40xx/004/build/tn40.c:3812:77: error: parameter name omitted
>  3812 | bdx_set_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)
>       |                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c:3812:111: error: parameter name omitted
>  3812 | bdx_set_coalesce(struct net_device *netdev, struct ethtool_coalesce *ecoal, struct kernel_ethtool_coalesce *, struct netlink_ext_ack *)
>       |                                                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c: At top level:
> /var/lib/dkms/tn40xx/004/build/tn40.c:3877:11: warning: ‘struct kernel_ethtool_ringparam’ declared inside parameter list will not be visible outside of this definition or declaration
>  3877 |    struct kernel_ethtool_ringparam *kernel_ering,
>       |           ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c:3898:11: warning: ‘struct kernel_ethtool_ringparam’ declared inside parameter list will not be visible outside of this definition or declaration
>  3898 |    struct kernel_ethtool_ringparam *kernel_ering,
>       |           ^~~~~~~~~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_ethtool_ops’:
> /var/lib/dkms/tn40xx/004/build/tn40.c:4113:20: error: initialization of ‘void (*)(struct net_device *, struct ethtool_ringparam *)’ from incompatible pointer type ‘void (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ [-Werror=incompatible-pointer-types]
>  4113 |   .get_ringparam = bdx_get_ringparam,
>       |                    ^~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c:4113:20: note: (near initialization for ‘bdx_ethtool_ops.get_ringparam’)
> /var/lib/dkms/tn40xx/004/build/tn40.c:4114:20: error: initialization of ‘int (*)(struct net_device *, struct ethtool_ringparam *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ [-Werror=incompatible-pointer-types]
>  4114 |   .set_ringparam = bdx_set_ringparam,
>       |                    ^~~~~~~~~~~~~~~~~
> /var/lib/dkms/tn40xx/004/build/tn40.c:4114:20: note: (near initialization for ‘bdx_ethtool_ops.set_ringparam’)
> cc1: some warnings being treated as errors
> make[1]: *** [scripts/Makefile.build:297: /var/lib/dkms/tn40xx/004/build/tn40.o] Błąd 1
> make[1]: *** Oczekiwanie na niezakończone zadania....
> make: *** [Makefile:1881: /var/lib/dkms/tn40xx/004/build] Błąd 2
> make: Opuszczenie katalogu '/usr/src/linux-headers-5.15.0-46-generic'
> 
Any ideas? Thanks!

cahz, 2022-09-09:
@worksasintended pointed out and fixed an issue with my patch (see #52). Does this solve your problem?

acooks, 2023-12-17:
bdx_[gs]et_ringparam for 5.15.x fixed in commit db4c74d74792634af in release/tn40xx-006

closed 2023-12-17 by acooks