acooks/tn40xx-driver #51: Issue with incompatible pointer types when compiling ubuntu 20.04 and LTS 22.04 LTS

scallipus, 2022-08-31:
make -C /lib/modules/5.15.0-46-generic/build M=/home/lloyd/tn40xx-driver-release-tn40xx-004 clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-46-generic'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-46-generic'
Building kernel 5.15.0-46-generic resume supported
make -C /lib/modules/5.15.0-46-generic/build M=/home/lloyd/tn40xx-driver-release-tn40xx-004 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.0-46-generic'
  CC [M]  /home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.o
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:3877:25: warning: ‘struct kernel_ethtool_ringparam’ declared inside parameter list will not be visible outside of this definition or declaration
 3877 |                  struct kernel_ethtool_ringparam *kernel_ering,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:3898:25: warning: ‘struct kernel_ethtool_ringparam’ declared inside parameter list will not be visible outside of this definition or declaration
 3898 |                  struct kernel_ethtool_ringparam *kernel_ering,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c: In function ‘bdx_ethtool_ops’:
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:4113:34: error: initialization of ‘void (*)(struct net_device *, struct ethtool_ringparam *)’ from incompatible pointer type ‘void (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ [-Werror=incompatible-pointer-types]
 4113 |                 .get_ringparam = bdx_get_ringparam,
      |                                  ^~~~~~~~~~~~~~~~~
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:4113:34: note: (near initialization for ‘bdx_ethtool_ops.get_ringparam’)
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:4114:34: error: initialization of ‘int (*)(struct net_device *, struct ethtool_ringparam *)’ from incompatible pointer type ‘int (*)(struct net_device *, struct ethtool_ringparam *, struct kernel_ethtool_ringparam *, struct netlink_ext_ack *)’ [-Werror=incompatible-pointer-types]
 4114 |                 .set_ringparam = bdx_set_ringparam,
      |                                  ^~~~~~~~~~~~~~~~~
/home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.c:4114:34: note: (near initialization for ‘bdx_ethtool_ops.set_ringparam’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:297: /home/lloyd/tn40xx-driver-release-tn40xx-004/tn40.o] Error 1
make[1]: *** [Makefile:1881: /home/lloyd/tn40xx-driver-release-tn40xx-004] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.0-46-generic'
make: *** [Makefile:147: all] Error 2

Does anyone have any idea how to fix this?

kalvdans, 2022-08-31:
Try https://github.com/acooks/tn40xx-driver/pull/48

scallipus, 2022-08-31:
totally resolved the issue thanks for that link I don't know why I did not see it before

acooks, 2023-12-16:
Fixed in release/tn40xx-006

closed 2023-12-16 by acooks