acooks/tn40xx-driver #52: Unable to install with Kernel 5.18.0-0.deb11.4

worksasintended, 2022-09-08:
```dkms install  -m tn40xx -v 004

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j40 KERNELRELEASE=5.18.0-0.deb11.4-amd64 -C /lib/modules/5.18.0-0.deb11.4-amd64/build M=/var/lib/dkms/tn40xx/004/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.18.0-0.deb11.4-amd64 (x86_64)
Consult /var/lib/dkms/tn40xx/004/build/make.log for more information.```

[make.log](https://github.com/acooks/tn40xx-driver/files/9530559/make.log)

worksasintended, 2022-09-08:
https://github.com/worksasintended/tn40xx-driver
this one works. Thanks to @cahz, you did 99.9% of the work. 
[The commit ](https://github.com/worksasintended/tn40xx-driver/commit/b11e0302cf0dbe1d699af0fb1a35a7b38004e767) only had a minor issue forgetting the variable names in the function definition

cahz, 2022-09-09:
Ah sorry and thanks for pointing it out. I apparently mixed up my patch management (I have a private fork for internal use) and based my public commit on [skwerlman's commit](https://github.com/skwerlman/tn40xx-driver/commit/a9f31bacee8b4304c24f24323fcd27d9cc8d3114), which fixed another issue.

devadattas, 2022-09-10:
Hi @worksasintended ,

Checking in if this issue got resolved? I am facing a similar issue here https://github.com/acooks/tn40xx-driver/issues/53

oktay89, 2023-05-19:
I can't install the drivers on my proxmox v7.4.3:
git clone -b release/tn40xx-004 https://github.com/acooks/tn40xx-driver.git /usr/src/tn40xx-004
dkms add -m tn40xx -v 004
dkms install -m tn40xx -v 004

This is the error i get:
Error! Bad return status for module build on kernel: 5.15.107-2-pve (x86_64)
Consult /var/lib/dkms/tn40xx/004/build/make.log for more information.

I already tried to clone the forks of the original repository like this:
git clone -b release/tn40xx-004 https://github.com/skwerlman/tn40xx-driver.git /usr/src/tn40xx-004-skwerlman
dkms add -m tn40xx -v 004-skwerlman
dkms install -m tn40xx -v 004-skwerlman

Also tried this repo: https://github.com/worksasintended/tn40xx-driver
Still getting the same error. I think my Proxmox just tries to install the original repo. Is some of my commands wrong?

acooks, 2023-12-17:
fixed in release/tn40xx-006

closed 2023-12-17 by acooks