acooks/tn40xx-driver #54: Unable to install driver on 5.19.0-76051900-generic

Siggzmeyer, 2022-10-02:
I can't seem to install the driver using after doing the git-clone, I am new to linux so I may be doing something wrong.

`root@pop-os:/usr/src# dkms install -m tn40xx -v 004
Creating symlink /var/lib/dkms/tn40xx/004/source -> /usr/src/tn40xx-004

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.19.0-76051900-generic -C /lib/modules/5.19.0-76051900-generic/build M=/var/lib/dkms/tn40xx/004/build KVERSION=5.19.0-76051900-generic...(bad exit status: 2)
ERROR (dkms apport): binary package for tn40xx: 004 not found
Error! Bad return status for module build on kernel: 5.19.0-76051900-generic (x86_64)
Consult /var/lib/dkms/tn40xx/004/build/make.log for more information.
`

cahz, 2022-10-04:
Please have a look into your log file `/var/lib/dkms/tn40xx/004/build/make.log`. Most likely it is the same issue as in some previous issues and can be fixed by using e.g. the develop branch of my [patched version](https://github.com/cahz/tn40xx-driver/tree/develop).

Siggzmeyer, 2022-10-05:
Hi, I can't seem to get your branch to work, I'm sure it's because I have little knowledge of how git clone works. The modified command I'm trying to run is.

`git clone -b release/tn40xx-004 https://github.com/worksasintended/tn40xx-driver.git /usr/src/tn40xx-004
dkms add -m tn40xx -v 004`

cahz, 2022-10-05:
This is another version, but should work as well. Your given command should actually be two commands. First `git ...` and then `dkms ...`. If this still results in an error, please provide the error message.

Siggzmeyer, 2022-10-06:
I think I got it to install, though the 10G NIC isn't showing up. 

`root@pop-os:/home/siggz# git clone -b develop https://github.com/cahz/tn40xx-driver.git /usr/src/tn40xx-004
Cloning into '/usr/src/tn40xx-004'...
remote: Enumerating objects: 659, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 659 (delta 4), reused 9 (delta 3), pack-reused 649
Receiving objects: 100% (659/659), 3.01 MiB | 6.50 MiB/s, done.
Resolving deltas: 100% (396/396), done.
root@pop-os:/home/siggz# dkms add -m tn40xx -v 004
Creating symlink /var/lib/dkms/tn40xx/004/source -> /usr/src/tn40xx-004
root@pop-os:/home/siggz# dkms install -m tn40xx -v 004

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
make -j32 KERNELRELEASE=5.19.0-76051900-generic -C /lib/modules/5.19.0-76051900-generic/build M=/var/lib/dkms/tn40xx/004/build KVERSION=5.19.0-76051900-generic...
cleaning build area...

tn40xx.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.19.0-76051900-generic/extra/

depmod...
`

cahz, 2022-10-07:
Do you get any mention of `tn40xx` in `dmesg`? And does the card show up on `lspci`?

Siggzmeyer, 2022-10-07:
I didn't find anything in dmesg, though I could have missed it. 

`lspci` shows the Ethernet controller listed.

2d:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter

Siggzmeyer, 2022-10-09:
I just bought a different card. I'll use it somewhere else.

closed 2022-10-09 by Siggzmeyer