acooks/tn40xx-driver #56: Error install on ml kernel 6.2.5 almalinux 9

rederik1986, 2023-03-15:
make: Entering directory '/usr/src/kernels/6.2.5-1.el9.elrepo.x86_64'                                                                                                                                                                     │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/tn40.o                                                                                                                                                                                           │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4.o                                                                                                                                                                                            │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/CX4_Linux.o                                                                                                                                                                                      │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy.o                                                                                                                                                                                     │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/QT2025_phy_Linux.o                                                                                                                                                                               │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy.o                                                                                                                                                                                   │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/TLK10232_phy_Linux.o                                                                                                                                                                             │
│  CC [M]  /var/lib/dkms/tn40xx/004/build/AQR105_phy.o                                                                                                                                                                                     │
│In file included from /var/lib/dkms/tn40xx/004/build/tn40.c:9:                                                                                                                                                                            │
│/var/lib/dkms/tn40xx/004/build/tn40.c: In function ‘bdx_probe’:                                                                                                                                                                           │
│/var/lib/dkms/tn40xx/004/build/tn40.h:885:19: error: too many arguments to function ‘netif_napi_add’                                                                                                                                      │
│  885 |                   netif_napi_add(dev, napi, poll, weight)                                                                                                                                                                         │
│      |                   ^~~~~~~~~~~~~~                                                                                                                                                                                                  │
│/var/lib/dkms/tn40xx/004/build/tn40.c:3577:9: note: in expansion of macro ‘LUXOR__NAPI_ADD’                                                                                                                                               │
│ 3577 |         LUXOR__NAPI_ADD(ndev, &priv->napi, bdx_poll, 64);                                                                                                                                                                         │
│      |         ^~~~~~~~~~~~~~~                                                                                                                                                                                                           │
│In file included from /var/lib/dkms/tn40xx/004/build/tn40.h:16,                                                                                                                                                                           │
│                 from /var/lib/dkms/tn40xx/004/build/tn40.c:9:                                                                                                                                                                            │
│./include/linux/netdevice.h:2586:1: note: declared here                                                                                                                                                                                   │
│ 2586 | netif_napi_add(struct net_device *dev, struct napi_struct *napi,                                                                                                                                                                  │
│      | ^~~~~~~~~~~~~~                                                                                                                                                                                                                    │
│make[1]: *** [scripts/Makefile.build:252: /var/lib/dkms/tn40xx/004/build/tn40.o] Error 1                                                                                                                                                  │
│make[1]: *** Waiting for unfinished jobs....                                                                                                                                                                                              │
│make: *** [Makefile:2021: /var/lib/dkms/tn40xx/004/build] Error 2                                                                                                                                                                         │
│make: Leaving directory '/usr/src/kernels/6.2.5-1.el9.elrepo.x86_64'

rederik1986, 2023-03-16:
I build develop. He gathered but the network card did not appear. dmesg after modprobe tn40xx
```
[   67.260013] tn40xx: loading out-of-tree module taints kernel.
[   67.260890] Tehuti Network Driver from https://github.com/acooks/tn40xx-driver, 004
[   67.260901] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG
```

cahz, 2023-03-16:
Which network card do you have? Could you please provide the output of `lspci | grep Tehuti`?

rederik1986, 2023-03-16:
1:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter

rederik1986, 2023-03-17:
I tryed, but no effect.
```
[root@localhost ~]# lspci -v -d 1fc9:
01:00.0 Ethernet controller: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter
        Subsystem: Tehuti Networks Ltd. Ethernet Adapter
        Flags: fast devsel, IRQ 255
        Memory at f0000000 (64-bit, prefetchable) [disabled] [size=64K]
        Capabilities: [50] MSI: Enable- Count=1/1 Maskable- 64bit+
        Capabilities: [78] Power Management version 3
        Capabilities: [80] Express Endpoint, MSI 00
        Capabilities: [100] Virtual Channel
```
```
[root@localhost ~]# echo -n 0000:1:00.0 > /sys/bus/pci/drivers/vfio-pci/unbind
-bash: /sys/bus/pci/drivers/vfio-pci/unbind: No such file or directory
```

cahz, 2023-03-17:
The print does not show a kernel module, so there is no other module binded to it.

Does your card use a Marvell PHY? Then you need to enable it when compiling.

rederik1986, 2023-03-17:
The network card has a Marvel chip. I assume this is it. How to enable its support when compiling?

cahz, 2023-03-17:
`make help` lists the options, so you probably either want to use `make MV88X3120=YES` or `make MV88X3310=YES`. It will require some header file which you can find in the old vendor-drop branches.

rederik1986, 2023-03-21:
I build module in dkms.
In make help default build phys = all
How to specify build option in dkms?

cahz, 2023-03-21:
Maybe add it directly to this line here: https://github.com/acooks/tn40xx-driver/blob/release/tn40xx-004/dkms.conf#L4

rederik1986, 2023-03-21:
I rebuild with edited dkms.conf
`MAKE[0]="make -C ${kernel_source_dir} M=${dkms_tree}/${PACKAGE_NAME}/${PACKAGE_VERSION}/build KVERSION=${kernelver} MV88X3310=YES"`
But whei i modprobe tn40xx i see in dmesg
```
[  685.142208] tn40xx: loading out-of-tree module taints kernel.
[  685.143114] Tehuti Network Driver from https://github.com/acooks/tn40xx-driver, 004
[  685.143125] Supported phys :    QT2025 TLK10232 AQR105 MUSTANG
```
Why marvell not in list of phys?

cahz, 2023-03-21:
I think you need to clean/remove the dkms module first