acooks/tn40xx-driver #64: tn40xx 0000:07:00.0 enp7s0: fw 0xe [  644.481235] BUG: unable to handle page fault for address: fffff64d83ac3c00 [  644.481247] #PF: supervisor read access in kernel mode

afrojewelz, 2024-01-08:
[image]

root@asrb55z:~# lspci -nnk -s 07:00
07:00.0 Ethernet controller [0200]: Tehuti Networks Ltd. TN9710P 10GBase-T/NBASE-T Ethernet Adapter [1fc9:4027]
        Subsystem: Tehuti Networks Ltd. Ethernet Adapter [1fc9:3015]
        Kernel driver in use: tn40xx
[image]

use dkms add -m tn40xx/006 && dkms install -m tn40xx/006 on latest proxmox which is kernel 6.5.11-7-pve

the compiled module can be load,but whenever memory is occupied enough or running stress mem will result BUG unable to handle page fault for address: xxxx
then RIP my applications. although it's not fatal but it's causes unstable to sustain file service
i had onboard 2.5G realtek NIC when i switch to this one ,i can run stress memory whenever i want and stable which won't result RIP or BUG or crash. so if any great man can look into codes and try to debug them. btw my phy is MV88X3310.so i add hdr manually.

acooks, 2024-01-08:
You could try the potential  fix in commit
493dcc7fc80ac41ad40dfe3feccf735102944801

afrojewelz, 2024-01-09:
nope, after applied two changed files tn40xx.c & tn40xx.h in dev branch which was update 3weeks ago.still triggers page crashed.
[image]
it won't run normal until me unload the module and reboot with clean kernel without tn40xx.ko