acooks/tn40xx-driver #71: PHY init failed

Austayo, 2024-05-22:
Howdy,

Most likely forgetting something but installed the driver, build was clean and ez, but modprode then it go "nah sorry mate but I am just gonna make it harder for you to run me".

[image]

Austayo, 2024-05-22:
Added side notes: PHY is QT2025 and it has worked before.

degorychev, 2024-05-22:
I confirm the problem. Apparently the problem occurs in the release/linux-6.7.y-1 branch.
I'm trying to run unsopported PHY in proxmox using this project. On the host (Linux 6.8.4-3) the card refuses to work (PHY init failed). 
However, if you install a Debian 12 virtual machine (Linux 6.1.0-21) with a forwarded network card, the release/linux-6.6.y-1 branch works well (as far as I understand, it was correct to use release/tn40xx-006, but I only understood this now lol)

Austayo, 2024-05-22:
[image]
Magic, it worked straight away. Also does anyone know how to manually set the nic speed? Want it to run at 1gbps fd for a transciever I have.

Austayo, 2024-05-23:
New issue, the card does not detect any SFP module at boot, need to manually reinstert it then it detects it.